mit-han-lab/bevfusion #288: ValueError: matrix contains invalid numeric entries

lukas-staecker, 2023-01-17:
Hi,

I'm trying to train the fusion model with the config you provide, starting from the pretrained lidar-only detection weights and the pretrained camera swint backbone.
However, after several hundred iterations, I always get an error in the Hungarian assigner:
ValueError: matrix contains invalid numeric entries

Have you encountered this problem?

kentang-mit, 2023-01-17:
I encountered that problem before. Typically it is caused by incorrect learning rate schedule. However, this never happened in ~100 runs with this codebase using the latest schedule we released. I would suggest you to first confirm that the latest codebase is used. If it is the case, you could also try out FP32 training in the 1st epoch and switch to FP16 in later stages to see whether this issue will go away. Typically the FP16 training dynamics in the first hundreds of iterations might not be very stable and it could also be the reason for nans in the Hungarian assignment matrices.

kentang-mit, 2023-03-30:
Closed due to inactivity. Please feel free to reopen if you feel it necessary.

closed 2023-03-30 by kentang-mit

1653042420, 2023-10-26:
@kentang-mit   hi ，I only have one GPU. If I use the latest released code, do I still need to adjust the learning rate to  avoid this issue？