mit-han-lab/bevfusion #289: about sync_bn  when training  the fusion model

Paul-Zhu, 2023-01-18:
hello,  i turn on the sync_bn when training the fusion model，but the Loss unchanged. what is the reason why you did not use SyncBN ? Is it because of the pytorch version not support or when use SyncBN not improve precision so did not use?

kentang-mit, 2023-01-20:
Good question. Do you mean that the model will not converge if you turn on SyncBN or the accuracy will not improve if you use SyncBN? According to my experience, SyncBN could be helpful for LiDAR-only detectors, but the improvement is marginal. Since most of the competitor methods cited in our paper did not turn on SyncBN, we did not use it for fair comparisons.

Paul-Zhu, 2023-01-22:
thank you for the  reply. i have solved the question why the model will not converge by updated the pytorch version. after that, i found the SyncBN result when training the fusion model Almost the same with turn off SyncBN.

closed 2023-03-30 by kentang-mit