mit-han-lab/bevfusion #292: About pretrained checkpoint

sunnyHelen, 2023-01-30:
Hi, Haotian,
Do you have the well-trained checkpoint of Bevfusion under other settings like larger image size (384×1056) and different backbone (R50 and R101)? I really need it. Could you please share them~
Thanks a lot.

kentang-mit, 2023-01-30:
Hi @sunnyHelen,

I've sent you an Email about it. We could discuss in the Email thread.

Best,
Haotian

closed 2023-01-30 by kentang-mit

yinjunbo, 2023-10-24:
Hi, @kentang-mit 

I'd like to know how to apply the larger image size(384×1056)? 
If it needs to re-train the image model, can you share your pre-traned image model then?
Btw, does this parameter`resize: [[0.38, 0.55], [0.48, 0.48]]`need to be changed?

Thanks in advance!

Terencedu, 2023-11-07:
Hi @sunnyHelen and @kentang-mit 

Can I also get the pretrained model of R50 or radar models, because I hope to train camera + radar model. 

My email is 1062824592@qq.com

Thank you so much!