mit-han-lab/bevfusion #293: question abou model config

alchemz, 2023-02-02:
Dear Author, in transfusion.py, there is no usage of `post_center_range` , but in centerpoint.py, there is.

could you elaborate why `post_center_range` is different from `point_cloud_range` in centerpoint ? 

`point_cloud_range = [-51.2, -51.2, -5.0, 51.2, 51.2, 3.0]`
`post_center_range=[-61.2, -61.2, -10.0, 61.2, 61.2, 10.0]`

In the centerpoint task head, `point_cloud_range` has been used to filter the prediction. But it does not seems to be necessary, considering that all points beyond `point_cloud_range` have been filtered for training.
https://github.com/mit-han-lab/bevfusion/blob/3d91980d21556bf697f51b717875d0a31e5be2dd/mmdet3d/models/heads/bbox/centerpoint.py#L860-L870

kentang-mit, 2023-02-06:
Hi @alchemz,

These parameters are taken from the original CenterPoint implementation. The purpose of selecting a slightly larger range is to allow predictions out of [-51.2, 51.2], but I don't think it really matters for the nuScenes metrics. Feel free to change it back to `point_cloud_range`.

Best,
Haotian

alchemz, 2023-02-08:
@kentang-mit Thank you. I agree.

closed 2023-02-08 by alchemz