mit-han-lab/bevfusion #294: about changing backbone

hollow-503, 2023-02-02:
@kentang-mit Thanks for your work!
Could you tell me how you modified the `pointpillars` to adapt to your coordinates?
As mentioned above, I want to change the backbone to [sstv2](https://github.com/tusen-ai/SST). However, it used `dynamic_voxelize` instead of `hard_voxelize`. Therefore, I used the voxelization implementation of `sstv2` repo. I checked the #287 and modified the `scatter` from
```
indices = this_coors[:, 3] * ny + this_coors[:, 2]
```
to
```
indices = this_coors[:, 2] * nx + this_coors[:, 3]
```
at the end of `sstv2` because your coordinate is `xyz`. But it seems there are something wrong. I got `0 mAP` after 1 epoch. Could you share me the experiece how you modified the pointpillars to adapt to your coordinates?

hollow-503, 2023-02-02:
My config is
```yaml
voxel_size: [0.5, 0.5, 8]
point_cloud_range: [-50, -50, -5, 50, 50, 3]

model:
  encoders:
    lidar:
      voxelize:
        point_cloud_range: ${point_cloud_range}
        voxel_size: ${voxel_size}
        max_voxels: [-1, -1]
        max_num_points: -1
        Voxelization: true

      voxel_encoder:
        type: DynamicVFE
        in_channels: 5
        feat_channels: [64, 128]
        with_distance: false
        voxel_size: ${voxel_size}
        with_cluster_center: true
        with_voxel_center: true
        point_cloud_range: ${point_cloud_range}
      
      middle_encoder:
        type: SSTInputLayerV2
        window_shape: [16, 16, 1]
        sparse_shape: [200, 200, 1]
        shuffle_voxels: true
        debug: true
        pos_temperature: 10000
        normalize_pos: false
        mute: true

      backbone:
        type: SSTv2
        d_model: [128, 128, 128, 128, 128, 128, 128, 128]
        nhead: [8, 8, 8, 8, 8, 8, 8, 8]
        num_blocks: 8
        dim_feedforward: [256, 256, 256, 256, 256, 256, 256, 256]
        output_shape: [200, 200]
        num_attached_conv: 3
        conv_in_channel: 128
        conv_out_channel: 128
        debug: true

  heads:
    object:
      train_cfg:
        grid_size: [1600, 1600, 1]
      test_cfg:
        grid_size: [1600, 1600, 1]
```
The coords of `voxelize`, `voxel_encoder`, `middle_encoder`, `backbone` are all zyx. I just modified the code of converting learned features from dense tensor to sparse pseudo image (bev feature).

alchemz, 2023-02-03:
https://github.com/mit-han-lab/bevfusion/blob/3d91980d21556bf697f51b717875d0a31e5be2dd/mmdet3d/models/backbones/pillar_encoder.py#L155-L162
@hollow-503 This is my understanding. Feel free to correct me if I am wrong. I think at pillar_encoder, for feature center adjustment. If you want to change everything back to zyx, you need to make another adjustment
-  coors[:, 1]*self.vx --> coors[:, 3]*self.vx
- because in xyz coordinates, coors[:, 1] is x; 
- in zyx coordinates, coors[:, 3] is x; 

Below is the default implementation from mmdet3d in zyx
https://github.com/open-mmlab/mmdetection3d/blob/47285b3f1e9dba358e98fcd12e523cfd0769c876/mmdet3d/models/voxel_encoders/pillar_encoder.py#L130-L139

hollow-503, 2023-02-03:
@alchemz thanks for your reply!
In [PointPillarsScatter](https://github.com/mit-han-lab/bevfusion/blob/main/mmdet3d/models/backbones/pillar_encoder.py#L223), author also modified from `zyx --> xyz`. Here is my understanding, the reason why author modified [pillar_encoder](https://github.com/mit-han-lab/bevfusion/blob/main/mmdet3d/models/backbones/pillar_encoder.py#L155-L162) is that author use his own voxelization implementation(xyz). However, if i use the voxelization implementation(`zyx`) of other backbone such as `sst`, I think I don't have to modify [pillar_encoder](https://github.com/mit-han-lab/bevfusion/blob/main/mmdet3d/models/backbones/pillar_encoder.py#L155-L162). All I need to do is to modify [PointPillarsScatter](https://github.com/mit-han-lab/bevfusion/blob/main/mmdet3d/models/backbones/pillar_encoder.py#L223).  Am I right? Is there something I ignored?

alchemz, 2023-02-03:
@hollow-503 Jointly the `pts_voxel_encoder`(`PillarFeatureNet`) and `pts_middle_encoder`(`PointPillarsScatter`) give us bev features. They need to use the same voxel coordinates.

1. If you intent is only adding a new 'pts_middle_encoder' to bevfusion, and keep 'pts_voxel_encoder' as-is. Then it only needs to modify the pts_middle_encoder to align with bevfusion's XYZ. 

2. If you are planning to use default mmdet3d implementation(like I am doing), in ZYX for voxelization, then there is no need to change anything in the code base, because everything has already been aligned with this new format of ZYX for voxelization.

3. If the idea is to bring bevfusion's code base to mmdet3d(voxelization in ZYX), then we want to make sure whatever brough from bevfusion aligns with mmdet3d(voxelization in ZYX).

The key thing is to be consistency to use one type of coordinates between different components.

hollow-503, 2023-02-04:
@alchemz Thanks for your help！I found that `transfusion` head also calculate loss with `xyz`.Therefore, I transpose the bev feature（the backbone final output with `zyx`），but I still got 0 mAP.Have you found any other things I should modify except for coordinates alignment?

alchemz, 2023-02-04:
@hollow-503 Great discovery. It is hard for me to make a recommendation without knowing your exact change in the config. I would suggest you printing out the tensor from each below component, have a side by side comparison between bevfusion-transfusion output tensor and your new implementation tensor, so that you can know exactly what was going wrong.
- pts_voxel_encoder 
- pts_middle_encoder
- pts_backbone
- pts_neck
- pts_bbox_head

kentang-mit, 2023-02-06:
Thanks for the discussions. I could confirm that everything in our codebase follows the `xyz` convention. A good way to check whether the coordinate system is messed up is to use `tools/visualize.py` to visualize the ground truth boxes (in both BEV and 2D images). This will help you determine whether the ground truth boxes are in the correct coordinate system.

hollow-503, 2023-02-07:
> Thanks for the discussions. I could confirm that everything in our codebase follows the `xyz` convention. A good way to check whether the coordinate system is messed up is to use [...]

Thanks for your suggestion！The last I  want to know is what is the difference between your pointpillars implementation and offical implementation except for the coordinate？（about voxelization，middle encoder and backbone）

kentang-mit, 2023-02-09:
I think the implementations should be the same. The resolution out of BEV projection should be 512x512, the final output BEV space is 128x128. We use two PFE layers, each with 64 output channels to match the results in the CenterPoint codebase from Yin et al.

alchemz, 2023-02-10:
Just out of curiosity, I have modified the bevfusion lidar only config to use pointpillar. 

- The model A trained with voxelnet achieved 0.5511 mAP/0.5628 NDS. 
- The model B trained with pointpillar achieved 0.4506 mAP/0.5068 NDS. 

There is a 10% performance drop in model B. @kentang-mit Could you help confirm if this aligns with your experiment as well? Thanks!

kentang-mit, 2023-02-13:
Hi @alchemz,

Yes, I also observe roughly a 10 mAP gap between the best PointPillars model and the best VoxelNet model.

Best,
Haotian

closed 2023-02-18 by hollow-503

BigGold0202, 2023-08-24:
> Just out of curiosity, I have modified the bevfusion lidar only config to use pointpillar.
>
>     * The model A trained with voxelnet achieved 0.5511 mAP/0.5628 NDS.
>
>     * The model B trained with [...]

hello, I'm trying to use pointpillars as the fusion lidar backbone, but the NDS of the fusion model drops rapidly. Do you have similar problems?

wyf0414, 2024-04-25:
Hi, @BigGold0202 

I'm also changing the lidar backbone to pointpillars. But I'm not familiar, just changed the config. But it's running wrong.

Can you share how to change the backbone?

Thanks a lot.

LINSJGM, 2024-05-15:
Hello, could you share your well-configured PointPillars backbone code? Thank you very much.

 @alchemz @BigGold0202