mit-han-lab/bevfusion #296: Probelm of Evaluation

lilkeker, 2023-02-03:
[image]
Hi ,Haotiain,I encountered this bug in the 7th epoch when training the model, but there is no problem in the previous few epochs. Do you know how to solve it?

kentang-mit, 2023-02-06:
If it is the case, typically the model diverges in epoch 7. Is it the case that you tried out learning rate schedules other than what is provided by us?

lilkeker, 2023-02-09:
> 如果是这种情况，通常模型会在第 7 个时期发散。您是否尝试了我们提供的学习率计划以外的学习率计划？
[image]
Hi ，Haotian，thank you for your reply,
I changed some parameters of the optimizer, I changed "lr" to 1.0e-4, the above problem still occurs, I don't know how to avoid this problem, by the way, before encountering this problem, I modified the  depth network IN view- Transform part, I want to retrain this model, should I try other learning rate plans?

kentang-mit, 2023-02-09:
Another thing you could try is to train the model in FP32. But this will incur much larger memory cost. Reducing the batch size might be necessary. It could also be helpful if you could share with us some of the modifications you made to the depth network.

closed 2023-03-30 by kentang-mit