mit-han-lab/bevfusion #299: Problem about training

LiuYQ99, 2023-02-07:
I tried to train the segmentation model, but I met a problem:
```
Traceback (most recent call last):
  File "tools/train.py", line 87, in <module>
    main()
  File "tools/train.py", line 76, in main
    train_model(
  File "/media/bit202/文件&程序等/Yingqun.Liu/bevfusion/mmdet3d/apis/train.py", line 49, in train_model
    model.cuda(),
  File "/home/bit202/anaconda3/envs/bevf/lib/python3.8/site-packages/torch/nn/modules/module.py", line 680, in cuda
    return self._apply(lambda t: t.cuda(device))
  File "/home/bit202/anaconda3/envs/bevf/lib/python3.8/site-packages/torch/nn/modules/module.py", line 570, in _apply
    module._apply(fn)
  File "/home/bit202/anaconda3/envs/bevf/lib/python3.8/site-packages/torch/nn/modules/module.py", line 570, in _apply
    module._apply(fn)
  File "/home/bit202/anaconda3/envs/bevf/lib/python3.8/site-packages/torch/nn/modules/module.py", line 570, in _apply
    module._apply(fn)
  [Previous line repeated 2 more times]
  File "/home/bit202/anaconda3/envs/bevf/lib/python3.8/site-packages/torch/nn/modules/module.py", line 593, in _apply
    param_applied = fn(param)
  File "/home/bit202/anaconda3/envs/bevf/lib/python3.8/site-packages/torch/nn/modules/module.py", line 680, in <lambda>
    return self._apply(lambda t: t.cuda(device))
RuntimeError: CUDA error: out of memory
CUDA kernel errors might be asynchronously reported at some other API call,so the stacktrace below might be incorrect.
For debugging consider passing CUDA_LAUNCH_BLOCKING=1.
--------------------------------------------------------------------------
Primary job  terminated normally, but 1 process returned
a non-zero exit code. Per user-direction, the job has been aborted.
--------------------------------------------------------------------------
--------------------------------------------------------------------------
mpirun detected that one or more processes exited with non-zero status, thus causing
the job to be terminated. The first process to do so was:

  Process name: [[39800,1],0]
  Exit code:    1
--------------------------------------------------------------------------

```
I really want to know how to solve it. I used two 3090 to train, and I have changed two files into:
```
extra_compile_args["nvcc"] = extra_args + [
            "-D__CUDA_NO_HALF_OPERATORS__",
            "-D__CUDA_NO_HALF_CONVERSIONS__",
            "-D__CUDA_NO_HALF2_OPERATORS__",
            # "-gencode=arch=compute_70,code=sm_70",
            # "-gencode=arch=compute_75,code=sm_75",
            # "-gencode=arch=compute_80,code=sm_80",
            "-gencode=arch=compute_86,code=sm_86",
        ]
        sources += sources_cuda
```
and
```
samples_per_gpu: 2
  workers_per_gpu: 2
```
Could someone help me? Thank you.

kentang-mit, 2023-02-09:
If you are training the fusion model, 24G memory on 3090 is not enough. We train on 48G A6000s. A new version of this codebase (in which we will replace the spconv backend) should be able to support batch_size=2 under 24G memory. Please stay tuned.

LiuYQ99, 2023-02-10:
Thank you for your reply!  I'm looking forward to the new version of this codebase.

kentang-mit, 2023-03-30:
Closed due to inactivity. Please feel free to reopen if you feel it necessary.

closed 2023-03-30 by kentang-mit

Fiorentinar, 2023-07-03:
> If you are training the fusion model, 24G memory on 3090 is not enough. We train on 48G A6000s. A new version of this codebase (in which we will replace the spconv backend) should be able to support batch_size=2 under 24G memory. Please stay tuned.

Thank you for your work! I'm wondering if **torch.cuda.amp** can help to save memory and maintain the results of the training