mit-han-lab/bevfusion #301: Error when evaluation

Laobyy, 2023-02-10:
When I run the evaluation in the docker container:
torchpack dist-run -np 1 python tools/test.py configs/nuscenes/det/transfusion/secfpn/camera+lidar/swint_v0p075/convfuser.yaml pretrained/bevfusion-det.pth --eval bbox
occurs this error:
TypeError: 'MultiPolygon' object is not iterable

Here are the details:
Original Traceback (most recent call last):
  File "/opt/conda/lib/python3.8/site-packages/torch/utils/data/_utils/worker.py", line 287, in _worker_loop
    data = fetcher.fetch(index)
  File "/opt/conda/lib/python3.8/site-packages/torch/utils/data/_utils/fetch.py", line 49, in fetch
    data = [self.dataset[idx] for idx in possibly_batched_index]
  File "/opt/conda/lib/python3.8/site-packages/torch/utils/data/_utils/fetch.py", line 49, in <listcomp>
    data = [self.dataset[idx] for idx in possibly_batched_index]
  File "/home/bevfusion/mmdet3d/datasets/custom_3d.py", line 291, in __getitem__
    return self.prepare_test_data(idx)
  File "/home/bevfusion/mmdet3d/datasets/custom_3d.py", line 182, in prepare_test_data
    example = self.pipeline(input_dict)
  File "/opt/conda/lib/python3.8/site-packages/mmdet/datasets/pipelines/compose.py", line 41, in __call__
    data = t(data)
  File "/home/bevfusion/mmdet3d/datasets/pipelines/loading.py", line 290, in __call__
    masks = self.maps[location].get_map_mask(
  File "/opt/conda/lib/python3.8/site-packages/nuscenes/map_expansion/map_api.py", line 392, in get_map_mask
    return self.explorer.get_map_mask(patch_box, patch_angle, layer_names=layer_names, canvas_size=canvas_size)
  File "/opt/conda/lib/python3.8/site-packages/nuscenes/map_expansion/map_api.py", line 870, in get_map_mask
    map_mask = self.map_geom_to_mask(map_geom, local_box, canvas_size)
  File "/opt/conda/lib/python3.8/site-packages/nuscenes/map_expansion/map_api.py", line 819, in map_geom_to_mask
    layer_mask = self._layer_geom_to_mask(layer_name, layer_geom, local_box, canvas_size)
  File "/opt/conda/lib/python3.8/site-packages/nuscenes/map_expansion/map_api.py", line 1803, in _layer_geom_to_mask
    return self._polygon_geom_to_mask(layer_geom, local_box, layer_name, canvas_size)
  File "/opt/conda/lib/python3.8/site-packages/nuscenes/map_expansion/map_api.py", line 1890, in _polygon_geom_to_mask
    map_mask = self.mask_for_polygons(new_polygon, map_mask)
  File "/opt/conda/lib/python3.8/site-packages/nuscenes/map_expansion/map_api.py", line 1823, in mask_for_polygons
    exteriors = [int_coords(poly.exterior.coords) for poly in polygons]
TypeError: 'MultiPolygon' object is not iterable
/opt/conda/lib/python3.8/site-packages/shapely/set_operations.py:133: RuntimeWarning: invalid value encountered in intersection
  return lib.intersection(a, b, **kwargs)
/opt/conda/lib/python3.8/site-packages/shapely/set_operations.py:133: RuntimeWarning: invalid value encountered in intersection
  return lib.intersection(a, b, **kwargs)

/opt/conda/lib/python3.8/site-packages/shapely/set_operations.py:133: RuntimeWarning: invalid value encountered in intersection
  return lib.intersection(a, b, **kwargs)
/opt/conda/lib/python3.8/site-packages/shapely/set_operations.py:133: RuntimeWarning: invalid value encountered in intersection
  return lib.intersection(a, b, **kwargs)
/opt/conda/lib/python3.8/site-packages/shapely/set_operations.py:133: RuntimeWarning: invalid value encountered in intersection
  return lib.intersection(a, b, **kwargs)
/opt/conda/lib/python3.8/site-packages/shapely/set_operations.py:133: RuntimeWarning: invalid value encountered in intersection
  return lib.intersection(a, b, **kwargs)
/opt/conda/lib/python3.8/site-packages/shapely/set_operations.py:133: RuntimeWarning: invalid value encountered in intersection
  return lib.intersection(a, b, **kwargs)
/opt/conda/lib/python3.8/site-packages/shapely/set_operations.py:133: RuntimeWarning: invalid value encountered in intersection
  return lib.intersection(a, b, **kwargs)
/opt/conda/lib/python3.8/site-packages/shapely/set_operations.py:133: RuntimeWarning: invalid value encountered in intersection
  return lib.intersection(a, b, **kwargs)
/opt/conda/lib/python3.8/site-packages/shapely/set_operations.py:133: RuntimeWarning: invalid value encountered in intersection
  return lib.intersection(a, b, **kwargs)
/opt/conda/lib/python3.8/site-packages/shapely/set_operations.py:133: RuntimeWarning: invalid value encountered in intersection
  return lib.intersection(a, b, **kwargs)
/opt/conda/lib/python3.8/site-packages/shapely/set_operations.py:133: RuntimeWarning: invalid value encountered in intersection
  return lib.intersection(a, b, **kwargs)
/opt/conda/lib/python3.8/site-packages/shapely/set_operations.py:133: RuntimeWarning: invalid value encountered in intersection
  return lib.intersection(a, b, **kwargs)
/opt/conda/lib/python3.8/site-packages/shapely/set_operations.py:133: RuntimeWarning: invalid value encountered in intersection
  return lib.intersection(a, b, **kwargs)
/opt/conda/lib/python3.8/site-packages/shapely/set_operations.py:133: RuntimeWarning: invalid value encountered in intersection
  return lib.intersection(a, b, **kwargs)
/opt/conda/lib/python3.8/site-packages/shapely/set_operations.py:133: RuntimeWarning: invalid value encountered in intersection
  return lib.intersection(a, b, **kwargs)
/opt/conda/lib/python3.8/site-packages/shapely/set_operations.py:133: RuntimeWarning: invalid value encountered in intersection
  return lib.intersection(a, b, **kwargs)
/opt/conda/lib/python3.8/site-packages/shapely/set_operations.py:133: RuntimeWarning: invalid value encountered in intersection
  return lib.intersection(a, b, **kwargs)
/opt/conda/lib/python3.8/site-packages/shapely/set_operations.py:133: RuntimeWarning: invalid value encountered in intersection
  return lib.intersection(a, b, **kwargs)
/opt/conda/lib/python3.8/site-packages/shapely/set_operations.py:133: RuntimeWarning: invalid value encountered in intersection
  return lib.intersection(a, b, **kwargs)
/opt/conda/lib/python3.8/site-packages/shapely/set_operations.py:133: RuntimeWarning: invalid value encountered in intersection
  return lib.intersection(a, b, **kwargs)
--------------------------------------------------------------------------
Primary job  terminated normally, but 1 process returned
a non-zero exit code. Per user-direction, the job has been aborted.
--------------------------------------------------------------------------
--------------------------------------------------------------------------
mpirun detected that one or more processes exited with non-zero status, thus causing
the job to be terminated. The first process to do so was:

  Process name: [[20835,1],0]
  Exit code:    1
--------------------------------------------------------------------------
I think the problem is not about the environment setup cuz I am using the Dockerfile to build up in the container.

ahnz7, 2023-02-12:
I meet the same problem

Laobyy, 2023-02-13:
> 

Are you also using the nuscenes mini dataset instead of the full dataset?

ahnz7, 2023-02-13:
Hmmm,no, I am using the full dataset. You mean the reason of this error is using mini dataset?

Laobyy, 2023-02-13:
> Hmmm,no, I am using the full dataset. You mean the reason of this error is using mini dataset?

No, I am also not sure.
Did you also meet this error?
TypeError: 'MultiPolygon' object is not iterable
I thought there might be some data error in the mini dataset, but if the full dataset also occurs this I think the reason is not about the data error.

ahnz7, 2023-02-13:
yes, I met this problem yesterday, and I could not find any methods to fix it.

Laobyy, 2023-02-13:
> yes, I met this problem yesterday, and I could not find any methods to fix it.

Hi Hanlin, I downgrade the shapely from 2.0 to 1.8 and the error solved.

ahnz7, 2023-02-13:
I will try it, thx

ahnz7, 2023-02-13:
It works dude!thx!> > yes, I met this problem yesterday, and I could not find any methods to fix it.
> 
> 
> 
> Hi Hanlin, I downgrade the shapely from 2.0 to 1.8 and the error solved.

closed 2023-03-30 by kentang-mit