mit-han-lab/bevfusion #304: Error about single gpu test

WYYAHYT, 2023-02-15:
**Error info:**
 ```
 File "tools/test.py", line 232, in <module>
    main()
  File "tools/test.py", line 198, in main
    outputs = single_gpu_test(model, data_loader)
  File "/data/projects/bevfusion/mmdet3d/apis/test.py", line 12, in single_gpu_test
    result = model(return_loss=False, rescale=True, **data)
  File "/data/anaconda3/envs/bevfusion/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1110, in _call_impl
    return forward_call(*input, **kwargs)
  File "/data/anaconda3/envs/bevfusion/lib/python3.8/site-packages/mmcv/parallel/data_parallel.py", line 50, in forward
    return super().forward(*inputs, **kwargs)
  File "/data/anaconda3/envs/bevfusion/lib/python3.8/site-packages/torch/nn/parallel/data_parallel.py", line 166, in forward
    return self.module(*inputs[0], **kwargs[0])
  File "/data/anaconda3/envs/bevfusion/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1110, in _call_impl
    return forward_call(*input, **kwargs)
  File "/data/anaconda3/envs/bevfusion/lib/python3.8/site-packages/mmcv/runner/fp16_utils.py", line 139, in new_func
    output = old_func(*new_args, **new_kwargs)
  File "/data/projects/bevfusion/mmdet3d/models/fusion_models/bevfusion.py", line 187, in forward
    outputs = self.forward_single(
  File "/data/anaconda3/envs/bevfusion/lib/python3.8/site-packages/mmcv/runner/fp16_utils.py", line 139, in new_func
    output = old_func(*new_args, **new_kwargs)
  File "/data/projects/bevfusion/mmdet3d/models/fusion_models/bevfusion.py", line 244, in forward_single
    feature = self.extract_lidar_features(points)
  File "/data/projects/bevfusion/mmdet3d/models/fusion_models/bevfusion.py", line 132, in extract_lidar_features
    x = self.encoders["lidar"]["backbone"](feats, coords, batch_size, sizes=sizes)
  File "/data/anaconda3/envs/bevfusion/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1110, in _call_impl
    return forward_call(*input, **kwargs)
  File "/data/anaconda3/envs/bevfusion/lib/python3.8/site-packages/mmcv/runner/fp16_utils.py", line 139, in new_func
    output = old_func(*new_args, **new_kwargs)
  File "/data/projects/bevfusion/mmdet3d/models/backbones/sparse_encoder.py", line 116, in forward
    x = self.conv_input(input_sp_tensor)
  File "/data/anaconda3/envs/bevfusion/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1110, in _call_impl
    return forward_call(*input, **kwargs)
  File "/data/projects/bevfusion/mmdet3d/ops/spconv/modules.py", line 132, in forward
    input = module(input)
  File "/data/anaconda3/envs/bevfusion/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1110, in _call_impl
    return forward_call(*input, **kwargs)
  File "/data/projects/bevfusion/mmdet3d/ops/spconv/conv.py", line 163, in forward
    outids, indice_pairs, indice_pair_num = ops.get_indice_pairs(
  File "/data/projects/bevfusion/mmdet3d/ops/spconv/ops.py", line 92, in get_indice_pairs
    return get_indice_pairs_func(
**RuntimeError: mmdet3d/ops/spconv/src/indice_cuda.cu 118
cuda execution failed with error 209**
```
**Platform:**
Jetson Orin with cuda 11.4, Jetpack 5.0.1

I tried to search for this error, but got no solution,
Really hope someone can help!

from-Pb2Ag, 2023-02-16:
Maybe the discuss in issue#194 will help you, that a single GPU may run OOM unless is configured properly.

WYYAHYT, 2023-02-16:
@from-Pb2Ag, Thanks for your advice.
But I attempted to monitor memory while running the program, and device memory was always enough(Jetson Orin 32GB for both GPU and CPU).

And I wonder if I can install `spconv2.0` to avoid using `spconv in mmcv`, cuz `spconv2.0 `needs less memory, so which version of mmdet3d bevfusion based?

from-Pb2Ag, 2023-02-17:
> @from-Pb2Ag, Thanks for your advice. But I attempted to monitor memory while running the program, and device memory was always enough(Jetson Orin 32GB for both GPU and CPU).
>
> And I wonder if I can [...]

In my environment, the mmdet3d is built from the scratch, which is specified in readme, so the very version is `0.0.0`. Perhaps the version of spconv matters .

kentang-mit, 2023-03-30:
The error should be caused by OOM. Notice that the current version of our codebase actually does not support spconv 2.x. We will replace the backend with a faster version of TorchSparse (it is around 1.2x faster than spconv on Orin) in the near future, which consumes less memory. I'll temporarily closing this issue and feel free to reopen if you have further questions.

closed 2023-03-30 by kentang-mit

SeungHwi0613, 2023-06-08:
which version of Pytorch you use? now i try with Orin too.