mit-han-lab/bevfusion #305: The gap of only-lidar model

hht1996ok, 2023-02-17:
Thank you for your contribution, When I trained the only lidar branch, I found that its accuracy could not reach the reported mAP/NDS 64.68/69.28. Is there any additional setting during the training of only-lidar branch?

hht1996ok, 2023-02-17:
Can you provide the weight of the only-lidar branch weight which trained in train+val set

kentang-mit, 2023-03-30:
Sorry I missed your issue previously. We did not keep the pretrained models on train+val but it's performance should be similar to TransFusion-L.

hht1996ok, 2023-04-06:
@kentang-mit Thank you for your reply, I have found a solution from the source code of transfusion. May I ask if you have found the reason for the low precision of the Transfusion branch？

kentang-mit, 2023-04-07:
One possible reason is that this version of codebase has incorrect settings for the center of gt boxes [here](https://github.com/mit-han-lab/bevfusion/blob/db75150717a9462cb60241e36ba28d65f6908607/mmdet3d/datasets/nuscenes_dataset.py#L321). You should change the center from (0.5, 0.5, 0) back to (0.5, 0.5, 0.5) (default in mmdet3d) to avoid problems in gt database construction.

qianqianqianqian, 2023-04-23:
> @kentang-mit Thank you for your reply, I have found a solution from the source code of transfusion. May I ask if you have found the reason for the low precision of the Transfusion branch？

I have met the same problem, are there some solutions?

hht1996ok, 2023-04-23:
> One possible reason is that this version of codebase has incorrect settings for the center of gt boxes [...]

Thank you for your reply, I will try it.

longsuo, 2023-04-24:
> One possible reason is that this version of codebase has incorrect settings for the center of gt boxes [...]

 Do I need to rerun the tools/create_data.py after change the center from (0.5, 0.5, 0) back to  (0.5, 0.5, 0.5) ? Looking forward to your response.

kentang-mit, 2023-05-02:
Yes, that is true. @longsuo

closed 2023-07-14 by zhijian-liu