mit-han-lab/bevfusion #306: about FPN

Monoclinic, 2023-02-17:
Hello, thanks for your excellent work.
I notice that BEVFusion-e contains models with different FPNs. I wonder if the main differences are in the architecture (for example, PANet, SecondFPN, Vanilla FPN, etc.) or the feature resolution (1/8, 1/16, ...), and how much improvement it could bring.

kentang-mit, 2023-02-17:
The main difference is in architecture. The improvement for the single model is not that large (I remember it is at most 0.2-0.3 mAP), but it does bring some diversity for model ensemble.

Monoclinic, 2023-02-20:
Many thanks for your reply.
I have one more question. If I want to train models with different voxel sizes, how to adjust "max_voxels" according to voxel size?
In the config the settings are as follows:
[0.075m] -> [120k, 160k]
[0.1m] -> [90k, 120k]
So I wonder if it is a good choice to simply make them reciprocal, for example
[0.05m] -> [180k, 240k]
[0.15m] -> [60k, 80k]

kentang-mit, 2023-02-24:
Yes I think these modifications make sense to me.

closed 2023-02-27 by Monoclinic