mit-han-lab/bevfusion #307: Epoch

autumn-wong, 2023-02-17:
Thank you for such great work. During the training, I encountered the following problem:
When I train with 2 GPUs, the number in the red box is 815, while the number in the red box is 544 when I train with 3 GPUs. 
Is there any reason behind this?
[image]
[image]

kentang-mit, 2023-02-17:
I think the behavior is expected. Basically we expect number of iterations x number of gpus to be a constant. Because number of iterations x number of gpus x batch size = dataset size.

autumn-wong, 2023-02-23:
Oh, I see. I used the nuscenes-mini dataset for the experiment. Because of the equipment, the batchsize can only be set to 1, so 544 * 3 * 1=815 * 2 * 1. Thank you!

closed 2023-02-23 by autumn-wong

South-River, 2023-03-11:
Hello, 
I also tried to train it on the nuscenes-mini dataset. But I only get 0.42 map and 0.48 NDS under default config. Far worse than the model in the paper. And I'm training the model with full dataset, but I have to wait 5 days.  
So I want to know what's your result?