mit-han-lab/bevfusion #308: gt bbox coordinate transformation

llq12, 2023-02-17:
Dear authors:
     You have mentioned gt bboxes are belong with global coordinate.For training,you transformed gt_bbox into lidar coordinate, but,i don't find related code.can you tell me the location about this part code.thanks!

kentang-mit, 2023-02-17:
Good question. Actually the gt bboxes stored in the info file are indeed in **LiDAR-centric** coordinate systems, but server submission requires the results to be in **global** coordinate system / world coordinate system. The conversion code from LiDAR-centric coordinate system to world coordinate system can be found [here](https://github.com/mit-han-lab/bevfusion/blob/db75150717a9462cb60241e36ba28d65f6908607/mmdet3d/datasets/nuscenes_dataset.py#L613).

llq12, 2023-02-18:
Is this part belong with test/val part?So, before training, when we get dataset's pkl file in tools/data_convert/nuscenes_convert.py#L249~251,gt_bbox coordinate is already in lidar-centric coordinate system.During training,we put camara,lidar and gt_bbox all data into lidar-centric coordinate system, when test or val,we convert the result to be in  global coordinate system / world coordinate system?

llq12, 2023-02-18:
[here](https://github.com/XuyangBai/TransFusion/blob/master/tools/data_converter/nuscenes_converter.py#:~:text=gt_boxes%20%3D%20np.concatenate(%5Blocs%2C%20dims%2C%20%2Drots%20%2D%20np.pi%20/%202%5D%2C%20axis%3D1))

kentang-mit, 2023-02-24:
Yes your understanding is exactly correct.

llq12, 2023-02-27:
> Yes your understanding is exactly correct.

Thank you for your response!

kentang-mit, 2023-03-06:
I'm closing this issue since it is resolved. Feel free to reopen if you have further questions.

closed 2023-03-06 by kentang-mit