mit-han-lab/bevfusion #311: Nuscenes train annotation without location key

rylynchen, 2023-02-20:
Hi, I found a problem in nuscenes_dataset.py, like:
```
    def get_data_info(self, index: int) -> Dict[str, Any]:
        info = self.data_infos[index]

        data = dict(
            token=info["token"],
            sample_idx=info['token'],
            lidar_path=info["lidar_path"],
            sweeps=info["sweeps"],
            timestamp=info["timestamp"],
            location=info["location"],
        )
```
But, in nuscenes_infos_train.pkl info values, without location.
Anyone met the same problem ?

kentang-mit, 2023-02-24:
Are you using the pkl file generated from mmdetection3d? If so, I would recommend you to either regenerate the info file using our codebase or remove the location term in the `data` dict. Our data preprocessing code does record `location` in the pkl, as indicated [here](https://github.com/mit-han-lab/bevfusion/blob/db75150717a9462cb60241e36ba28d65f6908607/tools/data_converter/nuscenes_converter.py#L203).

kentang-mit, 2023-03-30:
Closed due to inactivity. Please feel free to reopen if you feel it necessary.

closed 2023-03-30 by kentang-mit

rylynchen, 2023-05-06:
Thanks for your reply ! Can you show the way how to use nuscenes_converter.py ? I can't find anything about it in README.