mit-han-lab/bevfusion #312: Problems w evaluation in docker container

yukijx, 2023-02-20:
Hello, when I try to run the evaluation in the provided docker container with either one of these commands: 
(1) torchpack dist-run -np 8 python tools/test.py configs/nuscenes/det/transfusion/secfpn/camera+lidar/swint_v0p075/convfuser.yaml pretrained/bevfusion-det.pth --eval bbox 
(2) torchpack dist-run -np 8 python tools/test.py configs/nuscenes/seg/fusion-bev256d2-lss.yaml pretrained/bevfusion-seg.pth --eval map

I get the same error that the file **data/nuscenes/nuscenes_infos_val.pkl** is not found. I indeed don't have this file but am not sure how to get it or where to go from here. Do you have an idea? Thanks in advance. 

Traceback (most recent call last):
  File "/opt/conda/lib/python3.8/site-packages/mmcv/utils/registry.py", line 52, in build_from_cfg
    return obj_cls(**args)
  File "/home/bevfusion/mmdet3d/datasets/nuscenes_dataset.py", line 143, in __init__
    super().__init__(
  File "/home/bevfusion/mmdet3d/datasets/custom_3d.py", line 66, in __init__
    self.data_infos = self.load_annotations(self.ann_file)
  File "/home/bevfusion/mmdet3d/datasets/nuscenes_dataset.py", line 202, in load_annotations
    data = mmcv.load(ann_file)
  File "/opt/conda/lib/python3.8/site-packages/mmcv/fileio/io.py", line 60, in load
    with BytesIO(file_client.get(file)) as f:
  File "/opt/conda/lib/python3.8/site-packages/mmcv/fileio/file_client.py", line 992, in get
    return self.client.get(filepath)
  File "/opt/conda/lib/python3.8/site-packages/mmcv/fileio/file_client.py", line 517, in get
    with open(filepath, 'rb') as f:
FileNotFoundError: [Errno 2] No such file or directory: 'data/nuscenes/nuscenes_infos_val.pkl'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "tools/test.py", line 230, in <module>
    main()
  File "tools/test.py", line 169, in main
    dataset = build_dataset(cfg.data.test)
  File "/home/bevfusion/mmdet3d/datasets/builder.py", line 42, in build_dataset
    dataset = build_from_cfg(cfg, DATASETS, default_args)
  File "/opt/conda/lib/python3.8/site-packages/mmcv/utils/registry.py", line 55, in build_from_cfg
    raise type(e)(f'{obj_cls.__name__}: {e}')
FileNotFoundError: NuScenesDataset: [Errno 2] No such file or directory: 'data/nuscenes/nuscenes_infos_val.pkl'
--------------------------------------------------------------------------
Primary job  terminated normally, but 1 process returned
a non-zero exit code. Per user-direction, the job has been aborted.
--------------------------------------------------------------------------
--------------------------------------------------------------------------
mpirun detected that one or more processes exited with non-zero status, thus causing
the job to be terminated. The first process to do so was:

  Process name: [[15116,1],6]
  Exit code:    1
--------------------------------------------------------------------------

yukijx, 2023-02-20:
I am also getting this error as well it seems.

RuntimeError: CUDA error: invalid device ordinal
CUDA kernel errors might be asynchronously reported at some other API call,so the stacktrace below might be incorrect.
For debugging consider passing CUDA_LAUNCH_BLOCKING=1.

yukijx, 2023-02-21:
I realized my mistake, I missed some parts of the instructions.

closed 2023-02-21 by yukijx

hbc2020, 2023-07-16:
Hi,
When I ran the script: python tools/create_data.py nuscenes --root-path ./data/nuscenes --out-dir ./data/nuscenes --extra-tag nuscenes --version v1.0-mini, this error occured: FileNotFoundError: NuScenesDataset: [Errno 2] No such file or directory: './data/nuscenes/nuscenes_infos_train.pkl'.
 Could you tell me how did you solve this problem? Thank you very much!