mit-han-lab/bevfusion #320: “”height” represent the height of the top center of the object？

U201310646, 2023-02-23:
Hi,

Thanks for open-sourcing the awesome work! A question is bothering me: 
The output of TransfusionHead in the bevfusion model contains keywords such as center, **`height`**, dim, rot, etc.
During the inference phase the output will be executed by the **get_bboxes** function, as follows:
https://github.com/mit-han-lab/bevfusion/blob/db75150717a9462cb60241e36ba28d65f6908607/mmdet3d/models/fusion_models/bevfusion.py#L284-L285

In function **get_bboxes** the predicted bboxes will be decoded as follows:
https://github.com/mit-han-lab/bevfusion/blob/db75150717a9462cb60241e36ba28d65f6908607/mmdet3d/models/heads/bbox/transfusion.py#L741-L749

In function **decode** **`height`** first subtracted from dim[:2] by half
https://github.com/mit-han-lab/bevfusion/blob/db75150717a9462cb60241e36ba28d65f6908607/mmdet3d/core/bbox/coders/transfusion_bbox_coder.py#L67

In visulize.py  **`height`**  subtracted from dim[:2] by half again
https://github.com/mit-han-lab/bevfusion/blob/db75150717a9462cb60241e36ba28d65f6908607/tools/visualize.py#L118-L119

At last **`height`** is visualized as the center of the bottom surface with `visualize_camera` and `visualize_lidar`.
so the predicted “”height” in Header model represent the height of the top center of the object？ why so strange?

best wishes

kentang-mit, 2023-02-24:
I did some experiments according to another issue and found that we may want to change the origin [here](https://github.com/mit-han-lab/bevfusion/blob/db75150717a9462cb60241e36ba28d65f6908607/mmdet3d/datasets/nuscenes_dataset.py#L321) from (0.5, 0.5 0) to (0.5, 0.5, 0.5). i.e., make sure the box origin is in the center. I'll publish a PR later to introduce this fix. The current implementation is written such that the visualization on 2D images is correct.

U201310646, 2023-02-25:
yeah, the visualization is right but it conflicts with lidar data augmentation like this issue #280. Changing origin form (0.5,0.5,0) to (0.5, 0.5, 0.5) can resolve this conflict, but need to remove _height -= dim[:2]/2_ in visualize.py. Base my understanfing the pred_dict[" height "] of TransfusionHead represents the height of the geometric center after set origin as (0.5,0.5,0.5) , whereas before it represents the height of the top surface.

U201310646, 2023-02-27:
One more question:
during the training phase if the backbone of lidar and camera need to freeze or just end-to-end to fineturn all paramaters.
Looking forward to your response!

closed 2023-03-07 by U201310646

Jayden9912, 2023-08-07:
> I did some experiments according to another issue and found that we may want to change the origin [...]

I am able to reproduce the result using the repo code (i.e. using 0.5,0.5,0). In this case, is it right to conclude that the model learn to offset the bbox center to the bottom center?