mit-han-lab/bevfusion #321: Image pre-processing

autumn-wong, 2023-02-23:
Hello, thanks for your great work. 
In the process of reproducing the code, we finded that after the image pre-processing, such color images are obtained and they were fed to the SwinTransformer moudle as input to extract image features. (It should be the ImageNormalize function that does the change to color images) 
Is there any special reason for pre-processing the original image into such a color image? Maybe for Extracting edge? We think that this operation may damage the rich context from camera.

[image]

kentang-mit, 2023-02-24:
Actually the `ImageNormalize` function is not changing the colors of the original image. It just converts the 8-bit RGB values to a float tensor using standard z-score normalization (and the mean and std values are from ImageNet).

kentang-mit, 2023-03-30:
Closed due to inactivity. Please feel free to reopen if you feel it necessary.

closed 2023-03-30 by kentang-mit

hehern, 2023-12-15:
maybe you use the "cv2.imwrite" to save the img. PIL reads in the order of rgb and cv2 writes in the order of bgr