mit-han-lab/bevfusion #323: How to train with only two camera data?

xsxsmm, 2023-02-25:
Hi, if I only use front camera and back camera for training,  there will be no overlapping parts on the camera features, is bev-pooling still necessary?   If do this, how to process the lidar data, do I need to retain the point cloud data for the areas not covered by the image, or delete it?

kentang-mit, 2023-03-06:
Hi @xsxsmm,

It will still be necessary. The reason why we need BEV pooling is that the BEV grids might be large and there could be more than 1 point within each grid even if all cameras do not overlap with each other. For LiDAR, you may just keep all the points for the best performance.

Best,
Haotian

closed 2023-03-07 by xsxsmm

llsense, 2023-11-15:
> Hi, if I only use front camera and back camera for training, there will be no overlapping parts on the camera features, is bev-pooling still necessary? If do this, how to process the lidar data, do I need to retain the point cloud data for the areas not covered by the image, or delete it?

Hello, I have the same needs as you. Have you completed the training yet? How is the effect?