mit-han-lab/bevfusion #325: NuScnens dataset splitting

jiangchaokang, 2023-02-28:
Hello ~ Thank you for your work which has been very helpful to me! I would like to ask a question about data set splitting. I want to train the proposed network on both the training set and the validation set. But I noticed that the dataset created does not have the nuscenes_infos_trainval.pkl file. How can I do to perform the training on both training and validation sets?
I am looking forward to your reply. Good luck!

kentang-mit, 2023-03-06:
You may directly concatenate the lists in `nuscenes_infos_train.pkl` and `nuscenes_infos_val.pkl` to get the desired file. This is how we create trainval infos.

jiangchaokang, 2023-03-16:
Loading these two files (```nuscenes_infos_train.pkl``` & ```nuscenes_infos_val.pkl```) generates two dictionaries. I wonder how to merge the two dictionaries properly. Can you release this part of the code for me? Thank you very much!

closed 2023-03-19 by jiangchaokang