mit-han-lab/bevfusion #326: Config on waymo

minrui-hust, 2023-02-28:
I know you are do soming code cleanning work before release the config for waymo dataset. 
Before that be done, could you give some hint on how to tune on waymo dataset?
What changed compared to nuscenes?

kentang-mit, 2023-03-06:
We will release the code for Waymo in the next version of this codebase. The major difference from nuScenes detection experiments is that we use CenterPoint instead of TransFusion.

minrui-hust, 2023-03-20:
Thanks a lot.
I also found that transfusion head does not work well on waymo dataset, when I previously evaluating Transfusion.
I have reproduced Bevfusion on nuscenes with a similar result,  however when I try it on waymo dataset, with image backbone  swin-t pretrained on coco and centerpoint head, it seems no gain compared to raw centerpoint.  I guess pretraining image backbone on waymo is cirtical,  am I right?  What's more, what is your image augmentation setting in detail on waymo?

minrui-hust, 2023-03-21:
I try a depth modulation on the predicted weight of different depth(like the spatial modulated attention), and got considerable gain.
On nuscenes val with out tta or enssemble:
[image]
about 1mAP

kentang-mit, 2023-03-23:
Your new results sound very promising to me. 

For Waymo, we did pretraining on its own images dataset instead of COCO. Our configurations for the fusion model is similar to DeepFusion (CVPR'22), where a larger rotation augmentation on LiDAR could matter a lot.

minrui-hust, 2023-03-23:
OK, I will refer DeepFusion for more details

lcc815, 2023-05-05:
> We will release the code for Waymo in the next version of this codebase. The major difference from nuScenes detection experiments is that we use CenterPoint instead of TransFusion.

hi, do  you have any plan when will you release your next version code?

kentang-mit, 2023-05-22:
Hi @lcc815,

I am switching to a different research project recently so it is hard to give a concrete timeline. But we have someone working on code cleanning now and hopefully we can get it done asap.

Best,
Haotian

Provemj, 2023-08-07:
> We will release the code for Waymo in the next version of this codebase. The major difference from nuScenes detection experiments is that we use CenterPoint instead of TransFusion.

Sorry to bother, but when would this version be released?

closed 2024-05-04 by zhijian-liu