mit-han-lab/bevfusion #330: What is the stopping epoch number of the camera-only segmentation training?

TongNing-Albert, 2023-03-06:
[image]
I tested the camera-only segmentation training task on two 3090RTX GPUs and took 13 hours to finish 1 epoch and stop without interruption. How many epochs do we need to train the camera-only segmentation model? 
I thought it should need more epochs since you said in other issues that you took almost 1 day to train a single modality model on 8 GPUs.

kentang-mit, 2023-03-06:
Hi @TongNing-Albert,

The model will be trained for 20 epochs. We train the model on A6000 which is slightly faster than 3090.

Best,
Haotian

TongNing-Albert, 2023-03-08:
> 

Hello Haotian

So my training task stopped at the first epoch without any inference. Do you have any idea what is going on?

Best
Tong

kentang-mit, 2023-03-15:
I'm not exactly sure. You probably want to add a `-v` flag to the command like:

```bash
torchpack dist-run -np [num gpus] -v ...
```
to see the error messages.