mit-han-lab/bevfusion #331: how to make a custom dataset for bevfusion

zzyoazzy, 2023-03-07:
@zhijian-liu 
  Thanks for your amazing work!
  I wanna know how to make a custom dataset upon other scenes. Should I follow the format like Nuscenes? I also confused about what type of data I need to label. image with 3D bounding-box and Lidar cloud？Is it necessary to generate a semantic map？
Looking forward to your reply!

kentang-mit, 2023-03-15:
There is no need to follow the format of nuScenes. Technically you need to implement the data preprocessor (that will be called by `tools/create_data.py`), the data loader (like these files under `mmdet3d/datasets/`). 3D bounding boxes need to be labeled if you want to do detection tasks. Semantic map is not necessary unless you are interested in map segmentation tasks.

kentang-mit, 2023-03-30:
Closed due to inactivity. Please feel free to reopen if you feel it necessary.

Rlndinesh, 2023-08-10:
How many types of data loaders are there and what are they?