mit-han-lab/bevfusion #333: Well-trained model question

South-River, 2023-03-07:
Hi, Haotian,
Do you have the well-trained model of Bevfusion under nuScenes (both segmentation and detection)? Could you please share them~
Thank you very much.

kentang-mit, 2023-03-15:
Hi @South-River,

The checkpoints might be too large for me to send it via Email (all my Email accounts have a size limitation of ~50M). However, we provided a download script [here](https://github.com/mit-han-lab/bevfusion/blob/main/tools/download_pretrained.sh). Would you mind trying it out?

Best,
Haotian

South-River, 2023-03-15:
> Hi @South-River,
>
> The checkpoints might be too large for me to send it via Email (all my Email accounts have a size limitation of ~50M). However, we provided a download script [...]

Thank you! I'd try it later.

South-River, 2023-03-15:
> Hi @South-River,
>
> The checkpoints might be too large for me to send it via Email (all my Email accounts have a size limitation of ~50M). However, we provided a download script [...]

I'm sorry that I had see your script, but it seems there's only pretrained model I think? And my computation resources is not so enough for me to train such big model. So is it possible to share me the model trained completely (after using train.py) if it doesn't matter you? Thank you very much.

kentang-mit, 2023-03-23:
Actually there are also links to fully-trained model checkpoints (e.g. bevfusion-det.pth and bevfusion-seg.pth) in the script I mentioned.

closed 2023-03-30 by kentang-mit