mit-han-lab/bevfusion #338: Training experimental results error

haoxiaoshuai1024, 2023-03-09:
Thank you for your good work. I tried to reproduce your code. For the 3D object detection task on the nuscenes val set, my reappearance result was MAP: 67.43 nds: 70.95. Then the experimental result in the paper is MAP: 68.5 nds: 71.4. Reap results and papers and papers The difference is very large, can you share the configs.yaml file in your training? In addition, the ablation experiment 4 Object Detection task performance Map: 66.4, NDS: 69.5. What configuration is used for ablation experiments?

kentang-mit, 2023-03-15:
Hi Xiaoshuai,

I've replied to this thread in your Email so I'm closing it now.

Best,
Haotian

closed 2023-03-15 by kentang-mit

qiuxianzi, 2023-06-14:
Thank you for your wonderful work. Can you please send me a copy of the training configuration file. For multi-modal 3D object detection, I have made many attempts on our machine,but the best results we got on the validation set was: MAP:67.90   NDS:70.99，I can't always reach the result in the paper. My email:qiuxianzi666@gmail.com.
                                                                                                                                                       Thank you so much.