mit-han-lab/bevfusion #340: How do you pretrain swint on nuiamges?

hollow-503, 2023-03-12:
I have tried traing the fusion model without pretrain weights on nuimages and got about 1.0 NDS lower. If I use the pretrain weights, i can get the reported results(about 71 NDS). 
Therefore, I wonder how you pretrain your swint on nuimages. Did you follow image MAE or any other pretext tasks? Could you please share the paper or method you used?

kentang-mit, 2023-03-15:
Hi @hollow-503,

MAE is not used. You can find some scripts in the original mmdet3d codebase to pretrain Cascade MaskRCNN with a ResNet101 backbone. We just replace the ResNet101 backbone with SwinT, and no other special trick is used.

Best,
Haotian

hollow-503, 2023-03-17:
Thanks for your reply! I'll check it. Have a nice day :)

closed 2023-03-17 by hollow-503