mit-han-lab/bevfusion #341: RuntimeError

SMSajadi99, 2023-03-13:
I installed all the libraries and cuda and installed the code for setup, but after running it in the command line, it gives this error:

**RuntimeError: Error compiling objects for extension**

Of course, I will attach the entire code process.
Please guide me on how to fix it

`F:\A_Univercity\Term_10\Project2\BevFusion\bevfusion-main\bevfusion-main>python setup.py develop
running develop
running egg_info
writing mmdet3d.egg-info\PKG-INFO
writing dependency_links to mmdet3d.egg-info\dependency_links.txt
writing top-level names to mmdet3d.egg-info\top_level.txt
adding license file 'LICENSE' (matched pattern 'LICEN[CS]E*')
reading manifest file 'mmdet3d.egg-info\SOURCES.txt'
writing manifest file 'mmdet3d.egg-info\SOURCES.txt'
running build_ext
C:\Users\smsaj\AppData\Local\Programs\Python\Python38\lib\site-packages\torch\utils\cpp_extension.py:305: UserWarning: Error checking compiler version for cl: [WinError 2] The system cannot find the file specified
  warnings.warn(f'Error checking compiler version for {compiler}: {error}')
building 'mmdet3d.ops.spconv.sparse_conv_ext' extension
Emitting ninja build file F:\A_Univercity\Term_10\Project2\BevFusion\bevfusion-main\bevfusion-main\build\temp.win-amd64-3.8\Release\build.ninja...
Compiling objects...
Allowing ninja to set a default number of workers... (overridable by setting the environment variable MAX_JOBS=N)
[1/7] C:/Program Files/NVIDIA GPU Computing Toolkit/CUDA\v11.1\bin\nvcc --generate-dependencies-with-compile --dependency-output F:\A_Univercity\Term_10\Project2\BevFusion\bevfusion-main\bevfusion-main\build\temp.win-amd64-3.8\Release\mmdet3d\ops\spconv\src/reordering_cuda.obj.d --use-local-env -Xcompiler /MD -Xcompiler /wd4819 -Xcompiler /wd4251 -Xcompiler /wd4244 -Xcompiler /wd4267 -Xcompiler /wd4275 -Xcompiler /wd4018 -Xcompiler /wd4190 -Xcompiler /EHsc -Xcudafe --diag_suppress=base_class_has_different_dll_interface -Xcudafe --diag_suppress=field_without_dll_interface -Xcudafe --diag_suppress=dll_interface_conflict_none_assumed -Xcudafe --diag_suppress=dll_interface_conflict_dllexport_assumed -DWITH_CUDA -IF:\A_Univercity\Term_10\Project2\BevFusion\bevfusion-main\bevfusion-main\mmdet3d\ops\spconv\include -IC:\Users\smsaj\AppData\Local\Programs\Python\Python38\lib\site-packages\torch\include -IC:\Users\smsaj\AppData\Local\Programs\Python\Python38\lib\site-packages\torch\include\torch\csrc\api\include -IC:\Users\smsaj\AppData\Local\Programs\Python\Python38\lib\site-packages\torch\include\TH -IC:\Users\smsaj\AppData\Local\Programs\Python\Python38\lib\site-packages\torch\include\THC "-IC:/Program Files/NVIDIA GPU Computing Toolkit/CUDA\v11.1\include" -IC:\Users\smsaj\AppData\Local\Programs\Python\Python38\include -IC:\Users\smsaj\AppData\Local\Programs\Python\Python38\include "-IC:\Program Files (x86)\Microsoft Visual Studio\2022\BuildTools\VC\Tools\MSVC\14.35.32215\include" "-IC:\Program Files (x86)\Microsoft Visual Studio\2022\BuildTools\VC\Tools\MSVC\14.35.32215\ATLMFC\include" "-IC:\Program Files (x86)\Microsoft Visual Studio\2022\BuildTools\VC\Auxiliary\VS\include" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.22000.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\\include\10.0.22000.0\\um" "-IC:\Program Files (x86)\Windows Kits\10\\include\10.0.22000.0\\shared" "-IC:\Program Files (x86)\Windows Kits\10\\include\10.0.22000.0\\winrt" "-IC:\Program Files (x86)\Windows Kits\10\\include\10.0.22000.0\\cppwinrt" "-IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.8\include\um" -c F:\A_Univercity\Term_10\Project2\BevFusion\bevfusion-main\bevfusion-main\mmdet3d\ops\spconv\src\reordering_cuda.cu -o F:\A_Univercity\Term_10\Project2\BevFusion\bevfusion-main\bevfusion-main\build\temp.win-amd64-3.8\Release\mmdet3d\ops\spconv\src/reordering_cuda.obj -D__CUDA_NO_HALF_OPERATORS__ -D__CUDA_NO_HALF_CONVERSIONS__ -D__CUDA_NO_BFLOAT16_CONVERSIONS__ -D__CUDA_NO_HALF2_OPERATORS__ --expt-relaxed-constexpr -w -std=c++14 -D__CUDA_NO_HALF_OPERATORS__ -D__CUDA_NO_HALF_CONVERSIONS__ -D__CUDA_NO_HALF2_OPERATORS__ -gencode=arch=compute_70,code=sm_70 -gencode=arch=compute_75,code=sm_75 -gencode=arch=compute_80,code=sm_80 -gencode=arch=compute_86,code=sm_86 -DTORCH_API_INCLUDE_EXTENSION_H -DTORCH_EXTENSION_NAME=sparse_conv_ext -D_GLIBCXX_USE_CXX11_ABI=0
FAILED: F:/A_Univercity/Term_10/Project2/BevFusion/bevfusion-main/bevfusion-main/build/temp.win-amd64-3.8/Release/mmdet3d/ops/spconv/src/reordering_cuda.obj
C:/Program Files/NVIDIA GPU Computing Toolkit/CUDA\v11.1\bin\nvcc --generate-dependencies-with-compile --dependency-output F:\A_Univercity\Term_10\Project2\BevFusion\bevfusion-main\bevfusion-main\build\temp.win-amd64-3.8\Release\mmdet3d\ops\spconv\src/reordering_cuda.obj.d --use-local-env -Xcompiler /MD -Xcompiler /wd4819 -Xcompiler /wd4251 -Xcompiler /wd4244 -Xcompiler /wd4267 -Xcompiler /wd4275 -Xcompiler /wd4018 -Xcompiler /wd4190 -Xcompiler /EHsc -Xcudafe --diag_suppress=base_class_has_different_dll_interface -Xcudafe --diag_suppress=field_without_dll_interface -Xcudafe --diag_suppress=dll_interface_conflict_none_assumed -Xcudafe --diag_suppress=dll_interface_conflict_dllexport_assumed -DWITH_CUDA -IF:\A_Univercity\Term_10\Project2\BevFusion\bevfusion-main\bevfusion-main\mmdet3d\ops\spconv\include -IC:\Users\smsaj\AppData\Local\Programs\Python\Python38\lib\site-packages\torch\include -IC:\Users\smsaj\AppData\Local\Programs\Python\Python38\lib\site-packages\torch\include\torch\csrc\api\include -IC:\Users\smsaj\AppData\Local\Programs\Python\Python38\lib\site-packages\torch\include\TH -IC:\Users\smsaj\AppData\Local\Programs\Python\Python38\lib\site-packages\torch\include\THC "-IC:/Program Files/NVIDIA GPU Computing Toolkit/CUDA\v11.1\include" -IC:\Users\smsaj\AppData\Local\Programs\Python\Python38\include -IC:\Users\smsaj\AppData\Local\Programs\Python\Python38\include "-IC:\Program Files (x86)\Microsoft Visual Studio\2022\BuildTools\VC\Tools\MSVC\14.35.32215\include" "-IC:\Program Files (x86)\Microsoft Visual Studio\2022\BuildTools\VC\Tools\MSVC\14.35.32215\ATLMFC\include" "-IC:\Program Files (x86)\Microsoft Visual Studio\2022\BuildTools\VC\Auxiliary\VS\include" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.22000.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\\include\10.0.22000.0\\um" "-IC:\Program Files (x86)\Windows Kits\10\\include\10.0.22000.0\\shared" "-IC:\Program Files (x86)\Windows Kits\10\\include\10.0.22000.0\\winrt" "-IC:\Program Files (x86)\Windows Kits\10\\include\10.0.22000.0\\cppwinrt" "-IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.8\include\um" -c F:\A_Univercity\Term_10\Project2\BevFusion\bevfusion-main\bevfusion-main\mmdet3d\ops\spconv\src\reordering_cuda.cu -o F:\A_Univercity\Term_10\Project2\BevFusion\bevfusion-main\bevfusion-main\build\temp.win-amd64-3.8\Release\mmdet3d\ops\spconv\src/reordering_cuda.obj -D__CUDA_NO_HALF_OPERATORS__ -D__CUDA_NO_HALF_CONVERSIONS__ -D__CUDA_NO_BFLOAT16_CONVERSIONS__ -D__CUDA_NO_HALF2_OPERATORS__ --expt-relaxed-constexpr -w -std=c++14 -D__CUDA_NO_HALF_OPERATORS__ -D__CUDA_NO_HALF_CONVERSIONS__ -D__CUDA_NO_HALF2_OPERATORS__ -gencode=arch=compute_70,code=sm_70 -gencode=arch=compute_75,code=sm_75 -gencode=arch=compute_80,code=sm_80 -gencode=arch=compute_86,code=sm_86 -DTORCH_API_INCLUDE_EXTENSION_H -DTORCH_EXTENSION_NAME=sparse_conv_ext -D_GLIBCXX_USE_CXX11_ABI=0
CreateProcess failed: The system cannot find the file specified.
[2/7] C:/Program Files/NVIDIA GPU Computing Toolkit/CUDA\v11.1\bin\nvcc --generate-dependencies-with-compile --dependency-output F:\A_Univercity\Term_10\Project2\BevFusion\bevfusion-main\bevfusion-main\build\temp.win-amd64-3.8\Release\mmdet3d\ops\spconv\src/indice_cuda.obj.d --use-local-env -Xcompiler /MD -Xcompiler /wd4819 -Xcompiler /wd4251 -Xcompiler /wd4244 -Xcompiler /wd4267 -Xcompiler /wd4275 -Xcompiler /wd4018 -Xcompiler /wd4190 -Xcompiler /EHsc -Xcudafe --diag_suppress=base_class_has_different_dll_interface -Xcudafe --diag_suppress=field_without_dll_interface -Xcudafe --diag_suppress=dll_interface_conflict_none_assumed -Xcudafe --diag_suppress=dll_interface_conflict_dllexport_assumed -DWITH_CUDA -IF:\A_Univercity\Term_10\Project2\BevFusion\bevfusion-main\bevfusion-main\mmdet3d\ops\spconv\include -IC:\Users\smsaj\AppData\Local\Programs\Python\Python38\lib\site-packages\torch\include -IC:\Users\smsaj\AppData\Local\Programs\Python\Python38\lib\site-packages\torch\include\torch\csrc\api\include -IC:\Users\smsaj\AppData\Local\Programs\Python\Python38\lib\site-packages\torch\include\TH -IC:\Users\smsaj\AppData\L [... cut ...]
  File "C:\Users\smsaj\AppData\Local\Programs\Python\Python38\lib\site-packages\torch\utils\cpp_extension.py", line 682, in win_wrap_ninja_compile
    _write_ninja_file_and_compile_objects(
  File "C:\Users\smsaj\AppData\Local\Programs\Python\Python38\lib\site-packages\torch\utils\cpp_extension.py", line 1355, in _write_ninja_file_and_compile_objects
    _run_ninja_build(
  File "C:\Users\smsaj\AppData\Local\Programs\Python\Python38\lib\site-packages\torch\utils\cpp_extension.py", line 1682, in _run_ninja_build
    raise RuntimeError(message) from e
RuntimeError: Error compiling objects for extension`

SMSajadi99, 2023-03-27:
Hi @kentang-mit, I'm trying to get it up and running in Ubuntu.

Best regards

kentang-mit, 2023-03-30:
It seems to be a CUDA installation problem. Do we have more context about the error you met with? For example, the content in `/var/log/cuda-installer.log`. Probably you could try out alternative ways of CUDA installations (e.g. using .deb).

SMSajadi99, 2023-04-01:
Hi @kentang-mit, The problem I noticed is that it can't install Cuda 11.1 and 11.3 drivers at all, but it can install Cuda itself. I installed it with this `sudo apt install nvidia-driver-440` command, but it still had a problem and it doesn't show Cuda with `nvidia-smi ` command

SMSajadi99, 2023-04-01:
Hi @kentang-mit, I tried two methods, but it still can't install
https://developer.nvidia.com/cuda-11.3.0-download-archive?target_os=Linux&target_arch=x86_64&Distribution=Ubuntu&target_version=20.04&target_type=runfile_local
https://developer.nvidia.com/cuda-11.3.0-download-archive?target_os=Linux&target_arch=x86_64&Distribution=Ubuntu&target_version=20.04&target_type=deb_local

SMSajadi99, 2023-04-01:
I even went with this site and all its output was the same as this site, but the process failed for me in the last box.

https://www.if-not-true-then-false.com/2021/install-nvidia-cuda-toolkit-on-ubuntu-debian-linux-mint/

https://www.youtube.com/watch?v=EFPxYotTwDI

SMSajadi99, 2023-04-01:
I also downloaded a separate Nvidia driver based on my system

https://us.download.nvidia.com/XFree86/Linux-x86_64/530.41.03/NVIDIA-Linux-x86_64-530.41.03.run

SMSajadi99, 2023-04-01:
I am working on a virtual machine, can this cause this problem?

kentang-mit, 2023-04-07:
I'm not quite sure about the exact details of your installation process, but in my opinion you should not use `sudo apt-get install` to install the NVIDIA driver because the version might be quite old. In fact if it is possible I even recommend you to remove the old versions of the driver in order to avoid conflicts. For the installation process of CUDA toolkit using `.run`, I remember that it is possible for you to install both the driver and the toolkit. Therefore it should be unnecessary for you to separately install the driver. By the way, if your machine has GUI, I would also recommend you to have a look at `nouveau`-related problems. It might be necessary to disable it before your installation.

closed 2023-04-07 by SMSajadi99