mit-han-lab/bevfusion #342: Flip reflects on Intrinsic

BHC1205, 2023-03-14:
Hi,In transforms_3d.py,
why horizontal flip over point clouds would be [[1, 0, 0], [0, -1, 0], [0, 0, 1]] instead of [[-1, 0, 0], [0, 1, 0], [0, 0, 1]] on the transformation matrix, Is the flip of point clouds different from the flip of images?
https://github.com/mit-han-lab/bevfusion/blob/db75150717a9462cb60241e36ba28d65f6908607/mmdet3d/datasets/pipelines/transforms_3d.py#L254

kentang-mit, 2023-03-15:
If I remember it correctly, this is because we are using PIL to apply transformation on the images. In this case, the order of W and H are actually the exact opposite compared with X and Y in point clouds. So you may observe different transformation matrices.

BHC1205, 2023-03-16:
Can it be understood that the difference in the transformation matrix of flip is due to the difference in the definition of XY in the coordinate system？

kentang-mit, 2023-03-23:
Yes, you are correct.

closed 2023-03-30 by kentang-mit