mit-han-lab/bevfusion #345: Ground truth shift?

voegtlel, 2023-03-15:
Hi there!

I'm trying to understand why this change: https://github.com/mit-han-lab/bevfusion/blob/main/mmdet3d/datasets/nuscenes_dataset.py#L319-L322
was made. By the docs of nuscenes, their center is actually at `(0.5, 0.5, 0.5)`, but if you change it (as in the code) to `(0.5, 0.5, 0)`, it'll actually shift all the boxes by 50% upwards. Where do you unshift these for the submission? It seems that the whole network trains and outputs these shifted boxes.

Thanks!

kentang-mit, 2023-03-15:
Hi @voegtlel,

We did recently notice that this change can bring about unnecessary confusions and could impact the generated ground truth database. I'll fix this in the future. So I would recommend you to switch this line of code back to (0.5, 0.5, 0.5). If you apply this modification, then [this](https://github.com/mit-han-lab/bevfusion/blob/db75150717a9462cb60241e36ba28d65f6908607/tools/visualize.py#L99) line of code in the visualization script will be unnecessary.

Best,
Haotian

voegtlel, 2023-03-15:
Thanks for the reply, that helps!

closed 2023-03-15 by voegtlel

VeeranjaneyuluToka, 2023-06-19:
@voegtlel , I still get GT shift even with the above change, would you mind sharing exact changes if you solved the above issue?

voegtlel, 2023-06-20:
Hi @VeeranjaneyuluToka, we have further modified the source code to match our experiments, so I cannot really tell you quickly what we changed. Actually, we finally migrated to DeepInteraction repo (also public). There, the code is not a modified version of mmdet3d, but a plugin to mmdet3d, thus it is more convenient to see the changes. It's effectively very similar to bevfusion, except for the fusion transformer part and a few other additions. I strongly recommend looking at their source code as a comparison.

VeeranjaneyuluToka, 2023-06-21:
Hi @voegtlel , thanks for quick reply!

I have changed something as below to get the GT in correct position (without a shiftt), just let me know in case if you remember anything from this.

In this file (https://github.com/mit-han-lab/bevfusion/blob/main/mmdet3d/core/bbox/structures/lidar_box3d.py), corners are getting computed in corners function, it does as follows to compute corners.

1. create corners in ENU coordinate system based on width, length, and height of bbox
2. rotate using -yaw-np.pi/2 which means these corners are getting transformed to FLU from ENU
3. And then translate to center
In third step, (x,y,z) which is a center is still in ENU, that also needs to be in FLU, so when i do that change, it works fine.

Thanks for your reference to DeepInteraction repo, i will analyse that as well.

Plz let me know if you can recollect and did the same change as above.

voegtlel, 2023-06-21:
As said, it's a bit hard to separate those changes. Another note here: The heatmap is also rotated, but it's not trivial to change that. I fear that I cannot help you further with that, as it's too long ago to remember all the details. I hope DeepInteraction helps!