mit-han-lab/bevfusion #349: How to train my own data set on BEVfusion

Baron6977, 2023-03-21:
Thank you for the author's contribution ：）
 I want to train my own data set on BEVfusion. How can I do it?

kentang-mit, 2023-03-23:
Yes it is also possible, but probably you will have to implement the dataloaders for KITTI by yourself. There might be some references from mmdet3d that could be very helpful, but I am not very familiar with it.

SeungHwi0613, 2023-03-27:
I want to test and evaluate bevfusion with kitti dataset. then i don't have to change anything?  maybe it is little bit silly question for you... i just want to confirm it.

xxp912, 2023-04-07:
@Baron6977 
Hello,  I meet the same problem with you, may I ask if you were successful?

chenzb0712, 2023-08-10:
Hello, I meet the same problem with you, may I ask if you were successful?

Lcl159, 2023-09-26:
Hello, I meet the same problem with you, may I ask if you were successful?

liuyuansheng624, 2023-11-30:
> @Baron6977 Hello, I meet the same problem with you, may I ask if you were successful?

i can

liuyuansheng624, 2023-11-30:
> Hello, I meet the same problem with you, may I ask if you were successful?

sure

guoningwang, 2023-12-03:
@liuyuansheng624  你好，请问您是怎么训练kitti数据集的，方便分享一下吗？我怎么联系您？你可以guoningwang@163.com 联系我.

weilli, 2023-12-17:
> > @Baron6977 Hello, I meet the same problem with you, may I ask if you were successful?
> 
> i can

Hello! How about the performance in the KITTI (compared to the Voxel-RCNN)? Would you share the evaluation? Thanks!

liuyuansheng624, 2023-12-17:
you can &nbsp;wechat 13731082126 for me

发自我的iPhone

------------------ Original ------------------
From: weilli ***@***.***&gt;
Date: Sun,Dec 17,2023 10:08 PM
To: mit-han-lab/bevfusion ***@***.***&gt;
Cc: liuys ***@***.***&gt;, Mention ***@***.***&gt;
Subject: Re: [mit-han-lab/bevfusion]  can i train bevfusion with kitti dataset(Issue #349)

c-yyyy, 2024-04-09:
> Hello, I meet the same problem with you, may I ask if you were successful?

hello! Have you succeeded, please！

liuyuansheng624, 2024-04-09:
加我微信13731082126

发自我的iPhone

closed 2024-05-04 by zhijian-liu