mit-han-lab/bevfusion #350: something wrong when training bevfusion with nuscenes mini

YaqinLong, 2023-03-22:
When I tried to train bevfusion with nuscenes mini (v1.0 mini), I firstly prepared the dataset by running "python tools/create_data.py nuscenes --version v1.0-mini --root-path ./data/nuscenes --out-dir ./data/nuscenes --extra-tag nuscenes". Then, I trained the bevfusion and it went wrong with the following bug:"FileNotFoundError: NuscenesDataset: LoadBEVSegmentation: [Errno 2] No such file or directory:' data/nuscenes/maps/expansion/singapore-onenorth.json'". May I ask if I can train bevfusion with the mini dataset? And what's wrong with this bug?

closed 2023-03-22 by YaqinLong

hasaikeyQAQ, 2023-03-27:
I have encountered the same problem. Have you solved it?

YaqinLong, 2023-03-27:
> I have encountered the same problem. Have you solved it?

Yes， you can download the map expansion (v1.3), and put them into the "map".