mit-han-lab/bevfusion #351: RuntimeError: Dataloader worker (pid 15623) is killed by signaled: Killed

YaqinLong, 2023-03-22:
I set the samples_per_gpu:1 and workers_per_gpu:1,but still got the problem"RuntimeError: Dataloader worker (pid 15623) is killed by signaled: Killed". I wander it is caused by the limitation of GPU memory?

GDMG99, 2023-03-22:
Hi, 
Are you working on a docker container? If so, try running the image with the flag --ipc=host. I faced the same problem and that solved it. Hope it helps.

YaqinLong, 2023-03-23:
> Hi, Are you working on a docker container? If so, try running the image with the flag --ipc=host. I faced the same problem and that solved it. Hope it helps.

Thank you for your suggestion. I run bevfusion on Jetson Xavier NX where it can not use a docker container.

YaqinLong, 2023-03-29:
> Typically if a dataloader worker is killed it is more likely caused by CPU OOM. In your case @YaqinLong since NX has shared CPU and GPU memory and the current version of this codebase requires ~12GB memory to run (which I think is larger than the capacity of NX), so it could be difficult to run it on NX.

I got it. Thank you so much!

closed 2023-03-29 by YaqinLong