mit-han-lab/bevfusion #354: Testing with 2 GPUs doesn't finish running

ssuralcmu, 2023-03-23:
Command run-
torchpack dist-run -np 2 python tools/test.py configs/nuscenes/det/transfusion/secfpn/camera+lidar/swint_v0p075/convfuser.yaml pretrained/bevfusion-det.pth --eval bbox

I was able to run this with 2 2080Ti GPUs and get results. I upgraded to 2 A6000 GPUs today and ran the same thing. It hangs midway through execution. The same occurs with training with -np 2 argument. Testing works fine with np 1 perfectly fine. Do you know what might be going wrong? The CUDA version is still the same and I did not reinstall anything.

kentang-mit, 2023-03-30:
Did you have a chance to rerun `python setup.py develop`? It seems to me that the focal loss kernel is not compiled for your current device.

kentang-mit, 2023-04-07:
Hi @ssuralcmu,

Thanks a lot for your solution and explanation! I may want to highlight it to all other users that meet with the problem.

Best,
Haotian

ssuralcmu, 2023-04-07:
I tried installing mmcv from scratch as well by following their instructions. That didn’t resolved the problem  for me. Cuda did not get picked up during installation as there is something called an “MMCV Cuda compiler” that has to be installed and the additional steps added to the dockerfile were required.

closed 2023-04-07 by ssuralcmu