mit-han-lab/bevfusion #358: No protocol specified

gubei1998, 2023-03-27:
when I run the train,it will show like this:
     No protocol specified
     No protocol specified
     and then It will get stuck and not move, I I don't know what this problem is and how to solve it?

kentang-mit, 2023-03-30:
Hi @gubei1998,

Would you mind adding a `-v` flag to your training command and see if there are some more specific error messages? I vaguely remember that I saw the same warning during my training runs which did not end up being stuck.

Best,
Haotian

lixinghe1999, 2023-04-28:
@gubei1998 
From my experience, the reason can be the failure of multi-GPU processing, so please comment all the parallel-related code, for example dist.init() and then the test.py is ok.

liuyuansheng624, 2023-12-06:
> when I run the train,it will show like this: No protocol specified No protocol specified and then It will get stuck and not move, I I don't know what this problem is and how to solve it?

can you sloved it? can you tell me how to slove?

closed 2024-05-04 by zhijian-liu