mit-han-lab/bevfusion #362: size of points

gubei1998, 2023-03-28:
When I run bevfusion，it will Report an error：No protocol specified and I can't solve it.But I want to know What is the size of the points in vtransforms/base.py,because I use the image pipline in my net.And when I run my net,it will report an error :'RuntimeError: [enforce fail at CPUAllocator.cpp:68] . DefaultCPUAllocator: can't allocate memory: you tried to allocate 1126907510784 bytes. Error code 12 (Cannot allocate memory)
    depth = torch.zeros(batch_size, img.shape[1], 1, *self.image_size).to(',so I want to know the size of the points in 'batch_size = len(points)',then I can find what cause that error,I hope some of you who have run through bevfusion can help me.

kentang-mit, 2023-03-30:
Hi @gubei1998,

Judging from the error message you released, the allocator is trying to allocate around 1 TB of memory. For this specific depth variable, it's size should be `batch_size, 6, 1, 256, 704`. Therefore I do not expect it to take up such large amount of memory. Would you mind double-checking whether you have made any modifications to our codebase? 

Best,
Haotian

gubei1998, 2023-04-03:
Thank you for your answer!But when I use the vtransforms,I find that the size of points is (200000+,5),but not (batch_size,6,1,256,704),and I search the piplines for something about that,I can't find where you change the size of points,so I hope you can tell me about how to change the size about points,thank you very much!

gubei1998, 2023-04-03:
Thank you for your answer!But when I use the vtransforms,I find that the size of points is (200000+,5),the batch_size=len(points),so the batch_size = 200000+,then it will cost too much memory,so I think whether the size of points in my net is wrong and I search the piplines for something about that,I can't find where you change the size of points,so I hope you can tell me about how to change the size about points,thank you very much!

kentang-mit, 2023-04-07:
Thank you for the details. Our vtransform code actually assumes that the `points` are passed in as a list, not a `torch.Tensor`. Each tensor in the list should be corresponding to a LiDAR scan. So you might want to double check whether you changed the definition of `points` in your code (for example, you might have concatenated all the tensors in that list somewhere in your implementation).

closed 2023-07-14 by zhijian-liu