mit-han-lab/bevfusion #363: Output interrupt

hasaikeyQAQ, 2023-03-29:
When I use the instructions mentioned in readme.md for training. Command line instructions：`torchpack dist-run -np 2 python tools/train.py configs/nuscenes/seg/fusion-bev256d2-lss.yaml --model.encoders.camera.backbone.init_cfg.checkpoint pretrained/swint-nuimages-pretrained.pth`The output appears to be stuck. 
[image]
I have tried changing the Shapely library version to 1.8.0 and adding environment variables`CUDA_LAUNCH_BLOCKING=1`to force CUDA synchronization, but none of them has helped. （My current shapely version is 1.8.5）.I think this issue may be related to the following factors: My version of Torchpack is 0.3.1, which may have an incompatibility issue. In addition, I am using the Slurm job management system, which may cause the process to be killed.Do you have any other suggestions or solutions? If you need to provide more information, please let me know. Thank you for your help!

_Originally posted by @hasaikeyQAQ in https://github.com/mit-han-lab/bevfusion/issues/339#issuecomment-1488120007_

kentang-mit, 2023-03-30:
I've responded in issue 339. I'm sorry that we have not experimented with the slurm job managment system and we use MPI instead. Therefore it could be hard for me to help you debug this case.

hasaikeyQAQ, 2023-03-31:
Dear Haotian，

I have received your reply and thank you for your help.

Best regards

kentang-mit, 2023-04-07:
Sounds good! I'm closing this issue since it is resolved.

closed 2023-04-07 by kentang-mit