mit-han-lab/bevfusion #364: Something Wrong with vtransforms

YaqinLong, 2023-03-29:
I tried to export the camera-only model seperately and I changed export.py as follow:
`with torch.no_grad():
        for data in data_loader:

            img = data["img"].data[0]
            points = data["points"].data[0]
            camera2ego = data["camera2ego"].data[0]
            lidar2ego= data["lidar2ego"].data[0]
            lidar2camera = data["lidar2camera"].data[0]
            lidar2image = data["lidar2image"].data[0]
            camera_intrinsics = data["camera_intrinsics"].data[0]
            camera2lidar = data["camera2lidar"].data[0]
            img_aug_matrix = data["img_aug_matrix"].data[0]
            lidar_aug_matrix = data["lidar_aug_matrix"].data[0]
            metas = data["metas"].data[0]

            input_args = (img, points, camera2ego, lidar2ego, lidar2camera, lidar2image,
                          camera_intrinsics, camera2lidar, img_aug_matrix, lidar_aug_matrix)

            from functools import partial

            model.forward = partial(
                model.forward,
                metas=metas,
                rescale=True,
            )

            torch.onnx.export(
                model,
                input_args,
                "model.onnx",
                input_names=["input"],
                opset_version=13,
                do_constant_folding=True,
            )
`
But when I run the following code:
`python tools/export.py configs/nuscenes/det/centerhead/lssfpn/camera/256x704/swint/default.yaml pretrained/camera-only-det.pth`
it has bugs:
[image]
it seems that something wrong in the vtransform.
However, when I run test.py and train.py, it does not give an error.

kentang-mit, 2023-03-30:
I think it might be related to this [metas](https://github.com/mit-han-lab/bevfusion/blob/db75150717a9462cb60241e36ba28d65f6908607/mmdet3d/models/vtransforms/base.py#L222) field in the vtransform forward function. You may try to remove it from the argument list or anyway add it to other vtransform modules.

armbuster, 2023-03-31:
I was able to get past this issue by replacing the 0th item returned in [this](https://github.com/mit-han-lab/bevfusion/blob/main/mmdet3d/models/heads/bbox/transfusion.py#L843) list with ` rets[0][0]["bboxes"]`, it is returning a wrapper around a tensor to bbox coordinates that provides a bunch of accessors, and if you just return the tensor containing the coordinates without the wrapper around it nothing seems to break.

But the next issue I hit is:
'''
RuntimeError: Exporting the operator inverse to ONNX opset version 13 is not supported. Please feel free to request support or submit a pull request on PyTorch GitHub.
'''
ONNX supported this operator at one point but then removed it because its implementation had some problem
https://github.com/pytorch/pytorch/issues/64577

YaqinLong, 2023-04-02:
> I think it might be related to this [metas](https://github.com/mit-han-lab/bevfusion/blob/db75150717a9462cb60241e36ba28d65f6908607/mmdet3d/models/vtransforms/base.py#L222) field in the vtransform [...]

But when I add "metas = None" in the forward function,  I meet a new problem as follow:

[image]

kentang-mit, 2023-04-07:
Hi @armbuster,

Yes, this is the expected behavior. Inverse of a matrix is not natively supported in ONNX. To do the ONNX export of the view transformer, you have to assume that the `geom_feats` is precomputed (i.e. the `get_geometry` function is not executed on-the-fly). 

Best,
Haotian

kentang-mit, 2023-04-07:
Hi @YaqinLong,

Thanks for the followup. If you are doing the export for camera-only, I would recommend you to start by breaking down the models into three parts (camera encoder, vtransform and decoder). Then in this case, I don't think metas is required when you export vtransform. Again, you need to assume that `geom_feats` is precomputed, as described in my reply to @armbuster.

Note that we do have a internal version that supports end-to-end export, but the codebase structure is a bit different from the publicly-released one. The key idea is that you want to register the CUDA kernel for bev_pool using `register_custom_op_symbolic`. I'm not planning to release it recently due to limited bandwidth.

Best,
Haotian

YaqinLong, 2023-04-10:
> Hi @YaqinLong,
>
> Thanks for the followup. If you are doing the export for camera-only, I would recommend you to start by breaking down the models into three parts (camera encoder, vtransform and [...]

Hi,
Thank you for your replying. I will try what your say.

Best,

kentang-mit, 2023-05-02:
Hi @YaqinLong,

Sorry for the delayed response. I was very busy working on other projects recently. Regarding your problem, I think it is important to test whether it is possible to run the entire model without onnx export. Running these camera-only and LiDAR-only encoders separately should be fine. By the way, even if you can run the LiDAR encoder separately, I highly doubt whether it is practical to register it to ONNX because it contains a lot of sparse convolution layers that are not well-supported.

Best,
Haotian

closed 2023-07-15 by zhijian-liu

RangineniBalaji, 2024-07-17:
> I was able to get past this issue by replacing the 0th item returned in [this](https://github.com/mit-han-lab/bevfusion/blob/main/mmdet3d/models/heads/bbox/transfusion.py#L843) list with ` rets[0][0]["bboxes"]`, it is returning a wrapper around a tensor to bbox coordinates that provides a bunch of accessors, and if you just return the tensor containing the coordinates without the wrapper around it nothing seems to break. And then you can pass `metas=None` because that is the only place it is used.
> 
> But the next issue I hit is:
> 
> `RuntimeError: Exporting the operator inverse to ONNX opset version 13 is not supported. Please feel free to request support or submit a pull request on PyTorch GitHub.`
> 
> ONNX supported this operator at one point but then removed it because its implementation had some problem [pytorch/pytorch#64577](https://github.com/pytorch/pytorch/issues/64577)

@alexarmbr How do you managed to solve this issue.?