mit-han-lab/bevfusion #367: In the 3D detection task, map.json is needed?

JinSeoung-Oh, 2023-04-03:
Hi, first, thank for your wonderful work. 
I'm making a custom dataset and loader, and as you know map.json in nuScenes dataset is so difficult to make.
When I analyzed your code, I noticed map.json is not needed when 3D detection task.
It just need to make info.pkl file, maybe
It is true? My understanding is right? 
If not, I have to assume that map.json is given?

Thank for your reading and sorry about my pool english.
Thanks!

VeeranjaneyuluToka, 2023-04-04:
Hi, I am also training models with my own custom dataset without having map files for detection and i could train successfully. However i am yet to get good results.

JinSeoung-Oh, 2023-04-07:
> Hi, I am also training models with my own custom dataset without having map files for detection and i could train successfully. However i am yet to get good results.

Oh, thanks!
Actually, I am also face with that problem...
Training without map files is possible, but result is so pool..
Maybe map files is used for filtering detected box, I guess.. Maybe not
Anyway, I really appreciate your comment. 
Have a nice day!

Thanks!

closed 2023-04-07 by JinSeoung-Oh