mit-han-lab/bevfusion #368: About the downloading of pre-trained weights

LiewFeng, 2023-04-03:
Hi, @zhijian-liu , I use the link in download_pretrained.sh to download the weights in the browser, but obtain something strange and it doesn't donwload the weights. Could you give me some suggestions? Lots of thanks!
[image]

kentang-mit, 2023-04-07:
I don't think binary weights could be directly parsed in a text editor. You may need to use `torch.load`.

closed 2023-04-07 by kentang-mit

lee1221, 2023-04-15:
Hello, the official address for downloading the pre-training model of the project is invalid. If you download "bevfusion-det.pth" and "bevfusion-seg.pth", could you share it with my email address [185285659@qq.com], Wish you all the best in scientific research.