mit-han-lab/bevfusion #369: Cannot get 8.4 FPS on a RTX3090

armbuster, 2023-04-03:
My colleague wrote a loop which runs inference on [this](https://github.com/mit-han-lab/bevfusion/blob/main/configs/nuscenes/det/transfusion/secfpn/camera%2Blidar/swint_v0p075/convfuser.yaml) model as fast as possible. We only achieve ~2.5 FPS on an RTX3090, and inference on a single instance of the model uses about 8GB of GPU memory.

Could you provide some more context about the performance figure of 8.4 FPS reported in the paper? Did you do any TensorRT conversion, or use any code that is not in this repo? Did you use python/pytorch? Was there any parallelization? Any info that I could use to get from 2 FPS closer to 8 would be really helpful.

kentang-mit, 2023-04-07:
Hi @armbuster,

TensorRT is not used in our experiments. The way we get the speed is to sum up the runtime of all components in our model, with the time for input voxelization excluded. We don't count the time for voxelization because you can easily write a faster kernel to achieve this without storing all points in each grid to the DRAM (for example, using the dynamic voxelization implementation in CenterPoint official codebase). I guess the reason why you cannot see the 8.4FPS runtime is the slow voxelization.

Besides, the sparse convolution implementation used in this repo is a very slow one, it could be accelerated a lot. We are actually working on a new TorchSparse release that supports this new feature.

Best,
Haotian

armbuster, 2023-04-07:
Thanks Haotian for the context!

closed 2023-04-07 by armbuster