mit-han-lab/bevfusion #372: pretrained model downlad problem

SeungHwi0613, 2023-04-05:
Thanks for your great work....i have one problem to download pretrained model. i cannot access your mit files website. So when i try ./tools/download~.sh, i failed. There is any reason?

kentang-mit, 2023-04-07:
Hi @SeungHwi0613,

Would you mind having a try again? I just attempted to access the URLs to our pretrained checkpoints using cellular data (so that I am not in the school network), and I was able to download these files.

Best,
Haotian

XDU-WZ, 2023-04-09:
hi,when i download swint-nuimages-pretrained.pth,it has a problem as follow:
[image]]()

XDU-WZ, 2023-04-09:
hi,when i download swint-nuimages-pretrained.pth,it has a problem as follow:
[image]

SUZhaoyu, 2023-04-10:
same thing happened to me.

study0101, 2023-04-11:
same thing happened to me.  It cannot be opened in the browser either

liuliuliu11, 2023-04-14:
Hello, the official address for downloading the pre-training model of the project is invalid. If you have swint-nuimages-pretrained.pth, could you share it with my email address [(mailto:balms123456@gmail.com)], Wish you all the best in scientific research.

hubingqingorc, 2023-04-21:
I had the same issue and later using the chrono plugin in chrome downloaded fine

SeungHwi0613, 2023-04-21:
Thank you so much it is so helpful to me

flymin, 2023-04-26:
Hi @hubingqingorc @SeungHwi0613,
Could you please share the downloaded wights with me? My email address is rygao@cse.cuhk.edu.hk. I can provide the upload link on the cloud drive. Please get in touch with me by email.
I really appreciate any help you can provide.

huiyiygy, 2024-03-20:
HI, I have found a solution in #521. After replacement, the new download link is as follows
``` shell
mkdir pretrained &&
cd pretrained &&
wget http://hanlab18.mit.edu/projects/bevfusion/files/pretrained_updated/bevfusion-det.pth &&
wget http://hanlab18.mit.edu/projects/bevfusion/files/pretrained_updated/bevfusion-seg.pth &&
wget http://hanlab18.mit.edu/projects/bevfusion/files/pretrained/lidar-only-det.pth &&
wget http://hanlab18.mit.edu/projects/bevfusion/files/pretrained/lidar-only-seg.pth &&
wget http://hanlab18.mit.edu/projects/bevfusion/files/pretrained_updated/camera-only-det.pth &&
wget http://hanlab18.mit.edu/projects/bevfusion/files/pretrained_updated/camera-only-seg.pth &&
wget http://hanlab18.mit.edu/projects/bevfusion/files/pretrained_updated/swint-nuimages-pretrained.pth
```