mit-han-lab/bevfusion #373: question about len(points)

gubei1998, 2023-04-06:
I find that the type of points in data is datacontainer,and its size is (200000+,5),but when send points into the all net, I find that It will pass in the value of samples_per_gpu, at this time the value of len (points) will become the value of samples_per_gpu, for example,in vtransforms/base.py,I set the value of samples_per_gpu is 2,and then len(points) will change to 2 but not 200000+, I would like to know How are the points passed in by batch  , the whole bevfusion model data from the processing to pass into the network is how a process, I hope to get your answer, thank you very much!

kentang-mit, 2023-04-07:
Hi @gubei1998,

I replied in another issue but I can provide more details here. Basically it is our desired behavior that `len(points)=samples_per_gpu`. Actually `samples_per_gpu` means the batch size on each device. In fact these points are collated into batches in the voxelization function [here](https://github.com/mit-han-lab/bevfusion/blob/db75150717a9462cb60241e36ba28d65f6908607/mmdet3d/models/fusion_models/bevfusion.py#L137) in this codebase. After converting them into a batched representation, the points will be sent to a sparse convolution network and LiDAR features will be extracted.

Best,
Haotian

gubei1998, 2023-04-10:
What is the location of the voxelization function that processes points into batches, if I may ask?

VeeranjaneyuluToka, 2023-04-13:
@gubei1998 , @kentang-mit already highlighted a text with blue color, if you click on it, it will take you to voxelize function which is where it creates batches. Otherwise here in this file, you can find 
https://github.com/mit-han-lab/bevfusion/blob/db75150717a9462cb60241e36ba28d65f6908607/mmdet3d/models/fusion_models/bevfusion.py#L137

Hope this is what you are looking for.

gubei1998, 2023-04-14:
thank you very much！I really have not seen it！So sorry！

kentang-mit, 2023-05-02:
Thanks for the comments! I'm closing this issue since there is no more followups.

closed 2023-05-02 by kentang-mit