mit-han-lab/bevfusion #375: How to visualize the results

Hononon, 2023-04-09:
whne I run
```
torchpack dist-run -np 1 python tools/visualize.py configs/nuscenes/det/transfusion/secfpn/camera+lidar/swint_v0p075/convfuser.yaml
```
it's output is below
```
Traceback (most recent call last):
  File "tools/visualize.py", line 167, in <module>
    main()
  File "tools/visualize.py", line 83, in main
    for data in tqdm(dataflow):
TypeError: 'module' object is not callable
--------------------------------------------------------------------------
Primary job  terminated normally, but 1 process returned
a non-zero exit code. Per user-direction, the job has been aborted.
--------------------------------------------------------------------------
--------------------------------------------------------------------------
mpirun detected that one or more processes exited with non-zero status, thus causing
the job to be terminated. The first process to do so was:

  Process name: [[37400,1],0]
  Exit code:    1
```
how to fix it

yifengwang66, 2023-04-12:
I just meet the same problem, I modified the tools/visualize.py file like below:
```
 #from torchpack.utils.tqdm import tqdm
 from tqdm import tqdm
```
and then the problem solved.

Hononon, 2023-04-12:
> I just meet the same problem, I modified the tools/visualize.py file like below:
> 
> ```
>  #from torchpack.utils.tqdm import tqdm
>  from tqdm import tqdm
> ```
> 
> and then the problem solved.

I have solved it,too. I download `tqdm` from [here](https://github.com/zhijian-liu/torchpack/blob/master/torchpack/utils/tqdm.py).
And put it into `anaconda3/envs/bevfusion/lib/python3.8/site-packages/torchpack/utils`.Then the problem solved.

liuliuliu11, 2023-04-14:
Hello, the official address for downloading the pre-training model of the project is invalid. If you have swint-nuimages-pretrained.pth, could you share it with my email address [2894891376@qq.com]? I wish you all the best in scientific research.

closed 2023-04-19 by Hononon