mit-han-lab/bevfusion #377: Questions about the results

hasaikeyQAQ, 2023-04-11:
Thank the author for their excellent work. 
When I used the testing code provided in `readme.md` as follows:`torchpack dist-run -np 1 -v python tools/test.py configs/nuscenes/det/transfusion/secfpn/camera+lidar/swint_v0p075/convfuser.yaml /public/home/nngallery/test/bevfusion/pretrained/bevfusion-det.pth --eval bbox`, I obtained test results based on the mini dataset, as shown below:
[image]
Why are some category's AP results 0? 
Also, why did the test results of the three indicators AOE, AVE, and AAE in the last two categories show "nan"?

Rennylex, 2023-04-27:
> yes, I also obtain the same result ,but the dataset I used is nuscenes-mini. I guess it's possible that some samples in the dataset are missing data. Also, the AP of motorcycle and bicycle in my result are extremely low, even though I add the max_epochs to 40.

Hi Yifeng! Could you please share us about the modifications you've made to run the test/evaluation on the mini set? I can not preprocess the data correctly, and when I run the pretrained model with the provided commandline, I got stuck at "load checkpoint from http path: ......../swin_tiny_patch4_window7_224.pth"

lixinghe1999, 2023-04-28:
Also encounter the same problem on evaluation-mini, trying to test on full dataset now. 
Object Class    AP      ATE     ASE     AOE     AVE     AAE
car     0.928   0.173   0.157   0.100   0.117   0.074
truck   0.815   0.150   0.121   0.028   0.071   0.018
bus     0.994   0.164   0.077   0.026   0.427   0.400
trailer 0.000   1.000   1.000   1.000   1.000   1.000
construction_vehicle    0.000   1.000   1.000   1.000   1.000   1.000
pedestrian      0.926   0.118   0.257   0.378   0.217   0.125
motorcycle      0.734   0.195   0.272   0.361   0.053   0.000
bicycle 0.536   0.156   0.237   0.362   0.512   0.000
traffic_cone    0.888   0.083   0.343   nan     nan     nan
barrier 0.000   1.000   1.000   1.000   nan     nan

Interestingly, @hasaikeyQAQ and I get almost the same result, may I know whether you figure it out?

lixinghe1999, 2023-04-28:
> Hi Yifeng! Could you please share us about the modifications you've made to run the test/evaluation on the mini set? I can not preprocess the data correctly, and when I run the pretrained model with [...]

I can recommend my procedure:
1. download mini
2. MMdetection3D preprocessing - with mini argument
3. add map expansion
4. download pth and specify the right path

Rennylex, 2023-04-28:
> I can recommend my procedure:
>
> 1. download mini
> 2. MMdetection3D preprocessing - with mini argument
> 3. add map expansion
> 4. download pth and specify the right path

Thanks for the reply! I got that working earlier today. Your method is absolutely correct. The problem I've encountered is because I forgot to change the pkl file for v1.0-test back to the respective pkl file for mini.

Rennylex, 2023-04-28:
> Also encounter the same problem on evaluation-mini, trying to test on full dataset now. Object Class AP ATE ASE AOE AVE AAE car 0.928 0.173 0.157 0.100 0.117 0.074 truck 0.815 0.150 0.121 0.028 [...]

And I got the similar result. I'll stay tuned for any thoughts on this

lixinghe1999, 2023-04-30:
Dear all,

I just test the performance on full-nuscenes, almost the same as reported by the author.
From my understanding, the reason for lower performance on mini dataset is only randomness.
In other words, the model occasionally perform worse on those 2 scenes (val-mini)

kentang-mit, 2023-05-02:
Hi folks,

Sorry about missing the discussions. I was busy working on other projects recently. It is necessary to train the model on the entire training set following our recommended schedule. I did not try experiments on nuScenes-mini but as is mentioned by @lixinghe1999, the randomness in a small subset could usually explain the performance difference.

Best,
Haotian

closed 2023-07-14 by zhijian-liu