mit-han-lab/bevfusion #378: execute download_pretrained.sh Error

liuliuliu11, 2023-04-12:
Hi ,Thanks for your awesome paper and code!
When I run the script to download the model, I find that the web page is no longer accessible.
[image]

BigGold0202, 2023-04-16:
Same problem, the home page of this project is not feasible too

mohamed-ebbed, 2023-04-18:
I see that bevfusion.mit.edu is down

flymin, 2023-04-26:
The website seems still not working. Can anyone share the downloaded weights?

lcc815, 2023-04-27:
@flymin Could you please share the weights with us.
baidu net disk is ok.

lcc815, 2023-04-27:
> @flymin Could you please share the weights with us. baidu net disk is ok.

or my email is ccong.li@qq.com. Thanks a lot!

flymin, 2023-04-27:
> > @flymin Could you please share the weights with us. baidu net disk is ok.
> 
> or my email is [ccong.li@qq.com](mailto:ccong.li@qq.com). Thanks a lot!

I sent an email to you with the link, please check.

kentang-mit, 2023-05-02:
Thanks for reporting this problem. It seems to be a problem impacting users in many countries (e.g. in Europe and China). A VPN in the US will be helpful, but I will definitely discuss with my labmates to see whether we can find a solution without proxy servers.

kkrl1111, 2023-05-06:
I also need weight connections. Can you send me a copy? My email is iakrulan@wust.edu.cn ，Thank you！

Mandylove1993, 2023-05-06:
I also need weight connections too. Can you send me a copy? My email is zhuft161@163.com，Thank you！ @flymin

zhuliudanbao, 2023-05-06:
I also need weight connections too. Can you send me a copy? My email is charleyzhoucn@gmail.con，Thank you！ @flymin

BoomSky0416, 2023-05-09:
I also need weight connections too. Can you send me a copy? My email is [shoutian@umich.edu](mailto:shoutian@umich.edu)，Thank you！ @flymin

yukiser, 2023-05-18:
I also need weight connections too. Can you send me a copy? My email is 20173629@cqu.edu.cn，Thank you！ @flymin

kentang-mit, 2023-05-22:
Hi folks, 

Feel free to have a look at [this](https://drive.google.com/drive/folders/1Jru7VTfgvFF949DlP1oDfriP3wrmOd3c?usp=sharing) Google drive link and see if it works for you. I heard that our department is updating its HTTP server so there might be some impact on our project websites. We are still waiting for their feedback.

Best,
Haotian

Darkerok, 2023-09-26:
> > @flymin Could you please share the weights with us. baidu net disk is ok.
> 
> or my email is [ccong.li@qq.com](mailto:ccong.li@qq.com). Thanks a lot!

Could U plz share the weights with us? My email is 717376696@qq.com. Thanks a lot. We are still waiting for your feedback.

zhijian-liu, 2024-05-03:
Please update your local repository, as we have moved all the checkpoints to Dropbox.

closed 2024-05-03 by zhijian-liu