mit-han-lab/bevfusion #383: What is the function of "--bbox-classes" in 'tools/visualize.py'?

Hononon, 2023-04-14:
As to the topic,I don't understand what is the function of "--bbox-classes" in `tools/visualize.py`.
I set `parser.add_argument("--bbox-classes", nargs="+", type=int, default=['car','pedestrian'])`,the bbox in visualized picture is disappear.
And I set `parser.add_argument("--bbox-classes", nargs="+", type=int, default=2)`,the bbox in visualized picture is disappear,too.

kentang-mit, 2023-05-02:
Hi @Hononon,

Sorry for the delayed response because I was busy working on other projects recently. By specifying this field, you will be able to only visualize certain object classes instead of all classes. This line of code is actually not written by me, but `type=int` looks a bit strange to me. You may manually specify `args.bbox_classes=['car', 'pedestrian']` to see whether it works.

Best,
Haotian

Hononon, 2023-05-28:
Hi Haotian,
I will take note of your suggestion to manually specify the `args.bbox_classes` line of code with `['car', 'pedestrian']`. I will give it a try and see if it resolves the issue.

Thank you again for your input and for bringing this to my attention.

Best regards,
Hononon

closed 2023-07-14 by zhijian-liu