mit-han-lab/bevfusion #387: Questions about training results

tzy-0716, 2023-04-18:
Hi,
Thank you very much for sharing the paper and code!I reproduced your code, but the results were not ideal.I ran the code on RTX3080, and the code instructions are as follows:
torchpack dist-run -np 8 python tools/train.py configs/nuscenes/det/transfusion/secfpn/camera+lidar/swint_v0p075/convfuser.yaml

And my setting is samples_ per_ gpu: 1  workers_ per_ gpu: 1

The training results I received are as follows
mAP: 0.5847
mATE: 0.2932
mASE: 0.2724
mAOE: 0.3599
mAVE: 0.3110
mAAE: 0.1911
NDS: 0.6496

The results on the validation set are as follows
mAP: 0.5885                              
mATE: 0.2944
mASE: 0.2724
mAOE: 0.3571
mAVE: 0.3101
mAAE: 0.1905
NDS: 0.6518

This seems to be far from your result. Can you give me some advice? Is this related to me setting samples_per_gpu to 1?

Also, if I want to train the branches of radar and camera separately, how will I combine them? Can you give me some suggestions?

kentang-mit, 2023-05-02:
Hi @tzy-0716,

Sorry for the delayed response. I was busy working on other projects recently. I never tried batch=1 before, but I guess such a small batch size might have negative impact on the final performance because we have batch normalization layers in the BEV encoder and heads. BatchNorm2d layers typically expect slightly larger batch sizes to work. 

Radar-related models haven't been released in this codebase, but if you are talking about LiDAR, what we did is to pretrain the LiDAR-only model first, and then load its weights and jointly finetune it in the camera+LiDAR fusion model. Notice that we tried camera+radar before, it is better to train the fusion model end-to-end instead of doing pretraining-finetuning. 

Hope that my answer is helpful.

Best,
Haotian

lcc815, 2023-05-05:
> Hi @tzy-0716,
>
> Sorry for the delayed response. I was busy working on other projects recently. I never tried batch=1 before, but I guess such a small batch size might have negative impact on the [...]

Hi @kentang-mit ,
If you pretrained the Lidar-only model first, and  then load its weights and jointly finetune it in the camera+LiDAR fusion model, did you also pretrain the Camera-only model before training the fusion model?

Thanks

tzy-0716, 2023-05-06:
> Hi @kentang-mit , If you pretrained the Lidar-only model first, and then load its weights and jointly finetune it in the camera+LiDAR fusion model, did you also pretrain the Camera-only model before [...]

hi@kentang-mit

Thank you for taking the time out of your busy schedule to reply to my question. I will try again to train this model with a better server

tzy-0716, 2023-05-06:
> Hi @tzy-0716,
>
> Sorry for the delayed response. I was busy working on other projects recently. I never tried batch=1 before, but I guess such a small batch size might have negative impact on the [...]

hi @kentang-mit 

Thank you for taking the time out of your busy schedule to reply to my question. I will try again to train this model with a better server

closed 2023-07-14 by zhijian-liu

wulinaaa, 2023-07-24:
> 嗨，非常感谢您分享论文和代码！我复制了您的代码，但结果并不理想。我在RTX3080上运行了代码，代码说明如下：torchpack dist-run -np 8 python tools/train.py configs/nuscenes/det/transfusion/secfpn/camera+lidar/swint_v0p075/convfuser.yaml
> 
> 我的设置是samples_ per_ gpu：1 workers_ per_ gpu：1
> 
> 我收到的训练结果如下 mAP： 0.5847 mATE： 0.2932 mASE： 0.2724 mAOE： 0.3599 mAVE： 0.3110 mAAE： 0.1911 NDS： 0.6496
> 
> 验证集上的结果如下 mAP： 0.5885 mATE： 0.2944 mASE： 0.2724 mAOE： 0.3571 mAVE： 0.3101 mAAE： 0.1905 NDS： 0.6518
> 
> 这似乎与您的结果相去甚远。你能给我一些建议吗？这与我将samples_per_gpu设置为 1 有关吗？
> 
> 另外，如果我想分别训练雷达和相机的分支，我将如何组合它们？你能给我一些建议吗？

I want to know how can I set 'samples_ per_ gpu=1', I don't know which file it's in.

JunShao0104, 2023-08-16:
> Hi @tzy-0716,
>
> Sorry for the delayed response. I was busy working on other projects recently. I never tried batch=1 before, but I guess such a small batch size might have negative impact on the [...]

Hi Haotian, Could I ask about the results of your training on the Camera+Radar config? And the specific performance difference between end-to-end training and finetune training of CR model?

Terencedu, 2023-11-07:
Hi @tzy-0716 ,

Have you trained the Radar-related models? We don't have the pretrainded model of resnet50 and radar model, so I have no idea how to train CR model. Could you please share some ideas? Thanks a lot!

Best wishes.