mit-han-lab/bevfusion #388: About calibration

starnstar, 2023-04-18:
Hello, 
Thanks for your great work!
We have our own datasets. I wonder if the alignment between the lidar and the camera is not very good, for example, the intrinsics or extrinsics we calibrated are not very accurate, will it affect the performance of the BEVFusion network?
Thanks!

kentang-mit, 2023-05-02:
Hi @starnstar @VeeranjaneyuluToka,

Sorry for the delayed response. I would say yes, definitely calibration parameters could have some impact on the performance, but it is not that large. Previously I implemented a version of bevfusion which calculated the wrong `cam2lidar` and `lidar2cam` matrices, but it turns out that the final performance for the fusion model is only 0.1-0.2 mAP worse (than the correct implementation). So I guess the model is somehow robust to small misalignments in the calibration matrices.

Best,
Haotian

starnstar, 2023-05-04:
Hℹ @kentang-mit, 

Thanks for your reply!

closed 2023-05-04 by starnstar

nilesh-hampi, 2023-06-23:
Hi, 

I am trying to test the robustness against calibration. I applied 10° and 0.5m meters of de-calibration to cam2lidar and lidar2cam.  But I do not see any major change in the results. 

I am also not able to figure out where the cam2lidar and lidar2cam transformations are applied. I added de-calibration while loading the nuscenes data set. Also which is the common coordinate frame used? 

Thanks!