mit-han-lab/bevfusion #390: A trainning bug related to magma cuda

zjuliangxun, 2023-04-20:
When I was trying to finetune the segmentation head with the default configuration, where batch size equals one and workers_per_gpu equals 4, I came across with the following bug. The cuda memory is enough to finetune a head in the beginning, but later it fails.
```
python: /opt/conda/conda-bld/magma-cuda113_1619629459349/work/interface_cuda/interface.cpp:901: void magma_queue_create_from_cuda_internal(magma_device_t, cudaStream_t, cublasHandle_t, cusparseHandle_t, magma_queue**, const char*, const char*, int): Assertion `queue->dCarray__ != __null' failed.
```
The docker is built by the provided dockerfile.
In some issues, I found that it has something to do with the cuda memory. Therefore, I tried both on 3080Ti(12GB) and 3090Ti(24GB). However, the bug still happens after some iterations. Sometimes 500 iters, sometimes just 20 iters.
I doubt that this is a mmdet's bug, because it also happens in a lot of different projects. Could you please release your original docker image or tell me how to fix it?

kentang-mit, 2023-05-02:
Hi @zjuliangxun,

Sorry for the delayed response. I was busy working on other projects recently. I've never seen that error before, but if you were able to run the training job at least for some iterations and eventually it crashed, it seem to me that it could very likely be an OOM problem. I'm at least sure that 12GB memory cannot run training even for batch size = 1, but 24 GB should be enough for batch=1. You may run `watch gpustat` to check whether the GPU memory consumption grows very quickly during training. 

Best,
Haotian

zjuliangxun, 2023-05-03:
Hi, thanks a lot for your reply. I find that there is a pretty weird bug in the "spconv" operator, which leaks 8 GB cuda memory during inference/training. Therefore, I cannot even just finetune a single head on a 12GB machine because there is only 4 GB left. After I fix the bug by manually allocate global cuda memory, 12 GB GPU does become enough, but this magma bug occurs.  However, if I use a 24GB GPU, such as 3090, and don't fix the first memory leakage problem, everything works well.
I believe all these clues point to the "spconv" as the culprit. Just in case that you may want to fix it, using a higher-version spconv should work.

closed 2023-05-03 by zjuliangxun

Cancerce1l, 2023-07-26:
> 

Hi, I'm encountering the same problem with spconv. Can I ask how do you manually allocate global cuda memory, in order to fix this?

Cancerce1l, 2023-07-26:
> Hi, thanks a lot for your reply. I find that there is a pretty weird bug in the "spconv" operator, which leaks 8 GB cuda memory during inference/training. Therefore, I cannot even just finetune a [...]

Hi, I'm encountering the same problem with spconv. Can I ask how do you manually allocate global cuda memory, in order to fix this?

zjuliangxun, 2023-07-26:
Actually the only way to fix this bug is to use video cards with 24GB memory, rather than 12GB.
I find that although allocating global memory can have the model to run for a while, it will corrupts after several epochs. This time it throws a “maga” exception, which is still related to memory leakage.

从 Windows 版邮件发送

发件人: Cancerce1l
发送时间: 2023年7月26日 20:28
收件人: mit-han-lab/bevfusion
抄送: xunliang; State change
主题: Re: [mit-han-lab/bevfusion] A trainning bug related to magma cuda(Issue #390)

Hi, I'm encountering the same problem with spconv. Can I ask how do you manually allocate global cuda memory, in order to fix this?