mit-han-lab/bevfusion #391: Implementation of separate BEV encoder for each task

zehaoj, 2023-04-20:
Hi,

Thanks for this amazing work! I'm wondering how you have separate BEV encoder for each task so that the performance gap between joint training and single training can be further close? As stated in your paper. Thanks!!

kentang-mit, 2023-05-02:
Hi @zehaoj,

Sorry for the delayed response. A simple way is to replace `self.decoder` [here](https://github.com/mit-han-lab/bevfusion/blob/db75150717a9462cb60241e36ba28d65f6908607/mmdet3d/models/fusion_models/bevfusion.py#L63) to a dictionary of `self.decoders`, where `self.decoders["map"]` is for map segmentation and `self.decoders["det"]` is for detection. You will also need to have separate heads for both tasks, which is already natively supported.

Best,
Haotian

zehaoj, 2023-05-04:
Hi @kentang-mit 

Thanks for the reply! Amazing paper and repo!

closed 2023-05-04 by zehaoj