mit-han-lab/bevfusion #392: RuntimeError: CUDA error: no kernel image is available for execution on the device CUDA kernel errors might be asynchronously reported at some other API call,so the stacktrace below might be incorrect. For debugging consider passing CUDA_LAUNCH_BLOCKING=1

Aruen24, 2023-04-21:


kentang-mit, 2023-05-02:
Hi @Aruen24,

The error might be related to your GPU architecture. Please check whether your GPU architecture is included in [lines 22-25](https://github.com/mit-han-lab/bevfusion/blob/db75150717a9462cb60241e36ba28d65f6908607/setup.py#L22) in this file.

Best,
Haotian

closed 2023-07-14 by zhijian-liu

lrbob, 2023-10-17:
如何使用./vscode/launch.json debug train.py代码？

{
    "version": "0.2.0",
    "configurations": [
 
          {
                "name": "Python: Current File",
                "type": "python",
                "request": "launch",
                "console": "integratedTerminal",
                "args": [
                    "--np","2",
                    "tools/train.py",
                ],
                "env": {"CUDA_VISIBLE_DEVICES":"0,1"},
    
                 "program": "/opt/conda/lib/python3.8/site-packages/torch/distributed/run.py",
          }
        ]

}
这样设置以后运行debug还是提示KeyError: 'MASTER_HOST'

flstahl, 2023-12-01:
Hi @kentang-mit ,

I am currently experiencing a similar issue. I have checked the setup.py file and found that currently there is no mentioning of the Hopper architecture (sm_90). 
Is there any plan to include support for the new NVIDIA H100 GPUs? If not, could you help me understand how I could make it work on my own?

Thanks a lot!

EpicGilgamesh, 2024-01-05:
@flstahl have you managed to figure something out about this?

flstahl, 2024-01-05:
> @flstahl have you managed to figure something out about this?

Hi,
Yes, I have been able to get bevfusion to train on the Nvidia H100. I have commented in how to do this in a different issue.

Essentially, the important step is to include the specific gen code in the setup.py file. 

I found it works best when you delete all the lines mentioning other gen codes (e.g. sm_86). These are lines 22-25 if I am not mistaking. Afterwards you add one line with sm_90 and then build bevfusion by running the setup.py script.

Since I needed to get all of this running on torch 1.13.1, there were a couple of other steps needed but if you're fine with pytorch 2.x then you should be fine with the above step.

Lmk if this works for you!

Cheers

EpicGilgamesh, 2024-01-05:
@flstahl Yea I found your answer and it worked, but I'm still getting some strange errors.
Isn't recommended torch version 1.10.1?
I'm fighting with bevfusion for almost a month and I need it for my engineering thesis that will need to be done at the end of this month...
Did you managed to get whole bevfusion up and running?

flstahl, 2024-01-05:
@EpicGilgamesh not sure if torch 1.10 is recommended. The issue is that the Hopper Nvidia GPUs (H100) needs at least CUDA 11.8

This is causing some issues because not even pytorch 1.13.1 was officially shipped with CUDA 11.8. So in case you need some older pytorch versions you would have to build them from source. However, if torch 1.10 works with CUDA 11.8, I do not know. You would have to test that.

Cheers